yaqwsx/KiKit #716: KiCad 8.0.4 doesnt detect KiKit 1.5.0

elessargr, 2024-07-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

kikit, version 1.5.1

### KiCAD version

KiCad 8.0.4

### Operating system

Windows 10 Pro

### Description

Hello all,

I very recently installed and start learning how to use KiCad 8.0.3 installed. 
I managed yesterday to install and do the right steps to have KiKit installed and working to my computer.
Today after I did the upgrade from 8.0.3 to 8.0.4 KiKit stopped working. Tried clean re-install and nothing.
Tried full uninstall of the 8.0.4 and re-install along side with the plug in and pip KiKit install and again nothing.
Read the TS and steps that needs to be follow but no success. 

I will try to install 8.0.3 and see if this fixes the issue.
If anyone know how to fix this that would be amazing!

Couple screenshots
[image]
[image]

### Steps to Reproduce

Just by using KiCad 8.0.4

yaqwsx, 2024-07-18:
I believe this is a duplicate of #715. Could you please try the upstream version?

elessargr, 2024-07-18:
Hello Yaqwsx,
First of all I would like to say thank you! You have done an amazing job and script to help the electronics enthusiastic work on their favorite PCB easily. 

When you are saying "upstream version" can you please provide more clarity?

I can try once I confirm that the issue is with the 8.0.4 version. Now installing 8.0.3.

yaqwsx, 2024-07-18:
https://yaqwsx.github.io/KiKit/latest/installation/upgrading/

elessargr, 2024-07-18:
Confirmed in 8.0.3 works fine.
In 8.0.4 has the issue.
Following now the instruction from https://yaqwsx.github.io/KiKit/latest/installation/upgrading/ to see

[image]

yaqwsx, 2024-07-18:
Did you really install the upstream version? I believe this was fixed by eac56affe51dc7c99e426bc7b02e88b5e1d8b115 (which is in upstream and it is not released yet).

elessargr, 2024-07-18:
I think the upgrade was just to make sure I had the 1.5.1 KiKit version no?
I had it already installed. FYI on the PCM the version shown then is the 1.5.0 for some reason
```
C:\Users\user1\Documents\KiCad\8.0>kikit --version
kikit, version 1.5.1

C:\Users\user1\Documents\KiCad\8.0>pip install -U kikit
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: kikit in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (1.5.1)
Requirement already satisfied: click>=7.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (8.1.7)
Requirement already satisfied: commentjson>=0.9 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.0)
Requirement already satisfied: markdown2>=2.4 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.5.0)
Requirement already satisfied: numpy in c:\program files\kicad\8.0\bin\lib\site-packages (from kikit) (2.0.0)
Requirement already satisfied: pcbnewTransition<=0.5,>=0.4.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.4.1)
Requirement already satisfied: pybars3>=0.9 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.7)
Requirement already satisfied: shapely>=2.0.3 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.0.5)
Requirement already satisfied: solidpython>=1.1.2 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (1.1.3)
Requirement already satisfied: colorama in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from click>=7.1->kikit) (0.4.6)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from pybars3>=0.9->kikit) (0.5.1)
Requirement already satisfied: PrettyTable==0.7.2 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.7.2)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: ply<4.0,>=3.11 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.0.19)

C:\Users\user1\Documents\KiCad\8.0>kikit --version
kikit, version 1.5.1
```

Still same result. KiCad 8.0.4 doesnt see the KiKit :(

yaqwsx, 2024-07-18:
Released version = the version, that is installed/upgraded automatically (e.g., via `pip install -U kikit`)
Upstream version = unreleased, development version; installed e.g., via `pip install git+https://github.com/yaqwsx/KiKit@master`

Everything you show me is in the released version. I claim that the upstream version shouldn't suffer from this bug anymore and I ask you for verification. That is, please, install the upstream version and validate if the problem is gone or not.

yaqwsx, 2024-07-18:
Your system doesn't have git installed; please try `pip install https://github.com/yaqwsx/KiKit/archive/master.zip` instead

elessargr, 2024-07-18:
oh yes! Nice! Yes please add this command also to your webpage, it would be helpful for other people also I think.

Here the result:
```
C:\Users\user1\Documents\KiCad\8.0>pip install https://github.com/yaqwsx/KiKit/archive/master.zip
Collecting https://github.com/yaqwsx/KiKit/archive/master.zip
  Downloading https://github.com/yaqwsx/KiKit/archive/master.zip
     | 11.0 MB 31.1 MB/s 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in c:\program files\kicad\8.0\bin\lib\site-packages (from KiKit==0+unknown) (2.0.0)
Requirement already satisfied: pcbnewTransition<=0.5,>=0.4.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (0.4.1)
Requirement already satisfied: shapely>=2.0.3 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (2.0.5)
Requirement already satisfied: click>=7.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (8.1.7)
Requirement already satisfied: markdown2>=2.4 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (2.5.0)
Requirement already satisfied: pybars3>=0.9 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (0.9.7)
Requirement already satisfied: solidpython>=1.1.2 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (1.1.3)
Requirement already satisfied: commentjson>=0.9 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from KiKit==0+unknown) (0.9.0)
Requirement already satisfied: colorama in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from click>=7.1->KiKit==0+unknown) (0.4.6)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from commentjson>=0.9->KiKit==0+unknown) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from pybars3>=0.9->KiKit==0+unknown) (0.5.1)
Requirement already satisfied: PrettyTable==0.7.2 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->KiKit==0+unknown) (0.7.2)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->KiKit==0+unknown) (0.1)
Requirement already satisfied: ply<4.0,>=3.11 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->KiKit==0+unknown) (3.11)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in c:\users\user1\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->KiKit==0+unknown) (0.0.19)
Building wheels for collected packages: KiKit
  Building wheel for KiKit (setup.py) ... done
  Created wheel for KiKit: filename=KiKit-0+unknown-py3-none-any.whl size=252916 sha256=68d0dbf72f341a27b70d1654ef74f54451e76addf141f5ff4065002c87841fe7
  Stored in directory: C:\Users\user1\AppData\Local\Temp\pip-ephem-wheel-cache-dkwyirnu\wheels\c8\e9\96\97f984a9ebe3e705d949fd3cb8767e476963cf08a1b92c5f7b
Successfully built KiKit
Installing collected packages: KiKit
  Attempting uninstall: KiKit
    Found existing installation: KiKit 1.5.1
    Uninstalling KiKit-1.5.1:
      Successfully uninstalled KiKit-1.5.1
Successfully installed KiKit-0+unknown

C:\Users\user1\Documents\KiCad\8.0>kikit --version
kikit, version 0+unknown
```

YES! It seems that it worked! AWESOME!

[image]

yaqwsx, 2024-07-18:
Thanks for the confirmation that it works. There will be a release with this fix later today or tomorrow.

closed 2024-07-18 by yaqwsx