yaqwsx/KiKit #720: KiKit breaks because of NumPy version 2

viktak, 2024-07-23:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 11 Pro

### Description

Hello there and thanks for this plugin - been using it for quite some time.
Today I updated to the latest version (1.6.0) and ever since it wouldn't work with the latest version of KiCAD (8.0.4). 
I installed it using the usual method described on the home page:
```bash
D:\Documents\KiCAD\8.0>pip install kikit
Defaulting to user installation because normal site-packages is not writeable
Collecting kikit
  Downloading KiKit-1.6.0-py3-none-any.whl.metadata (5.7 kB)
Requirement already satisfied: click>=7.1 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (8.1.7)
Requirement already satisfied: commentjson>=0.9 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.0)
Requirement already satisfied: markdown2>=2.4 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.4.13)
Requirement already satisfied: numpy in c:\program files\kicad\8.0\bin\lib\site-packages (from kikit) (2.0.0)
Collecting pcbnewTransition<=0.5,>=0.4.1 (from kikit)
  Downloading pcbnewTransition-0.4.1-py3-none-any.whl.metadata (1.1 kB)
Requirement already satisfied: pybars3>=0.9 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.7)
Requirement already satisfied: shapely>=2.0.3 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.0.3)
Requirement already satisfied: solidpython>=1.1.2 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (1.1.3)
Requirement already satisfied: colorama in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from click>=7.1->kikit) (0.4.6)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from pybars3>=0.9->kikit) (0.5.1)
Collecting numpy (from kikit)
  Using cached numpy-1.26.4-cp311-cp311-win_amd64.whl.metadata (61 kB)
Requirement already satisfied: PrettyTable==0.7.2 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.7.2)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: ply<4.0,>=3.11 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in d:\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.0.19)
Downloading KiKit-1.6.0-py3-none-any.whl (252 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 252.9/252.9 kB 3.9 MB/s eta 0:00:00
Downloading pcbnewTransition-0.4.1-py3-none-any.whl (6.9 kB)
Using cached numpy-1.26.4-cp311-cp311-win_amd64.whl (15.8 MB)
Installing collected packages: pcbnewTransition, numpy, kikit
  Attempting uninstall: pcbnewTransition
    Found existing installation: pcbnewTransition 0.4.0
    Uninstalling pcbnewTransition-0.4.0:
      Successfully uninstalled pcbnewTransition-0.4.0
Successfully installed kikit-1.6.0 numpy-1.26.4 pcbnewTransition-0.4.1
```

When I test the installation, I get the following error:
```bash
D:\Documents\KiCAD\8.0>kikit --help

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\Scripts\kikit.exe\__main__.py", line 4, in <module>
    from kikit.ui import cli
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 6, in <module>
    from kikit.common import normalize
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\common.py", line 13, in <module>
    from shapely.geometry import LinearRing
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\shapely\__init__.py", line 13, in <module>
    from shapely.lib import GEOSException  # NOQA
AttributeError: _ARRAY_API not found
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\Scripts\kikit.exe\__main__.py", line 4, in <module>
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 6, in <module>
    from kikit.common import normalize
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\common.py", line 13, in <module>
    from shapely.geometry import LinearRing
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\shapely\__init__.py", line 13, in <module>
    from shapely.lib import GEOSException  # NOQA
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: numpy.core.multiarray failed to import
```

### Steps to Reproduce

Just install it using the instructions above.

yaqwsx, 2024-07-23:
For reasons unobvious to me, the **path** release of KiCAD 8.0.4 upgraded a major dependency (Numby) to a new **major** release. Also, it seems that KiCAD doesn't handle this upgrade well on itself and this errors appears when you just upgraded KiCAD. Based on [this answer], you should uninstall KiCAD completely and install it from scratch.

viktak, 2024-07-23:
@yaqwsx based on WHAT answer? :)

yaqwsx, 2024-07-23:
Ah, I fixed the link.

viktak, 2024-07-23:
I'm afraid that did not fix the issue. I just removed then reinstalled 8.0.4, and having the same issue.
Would it be possible to compile KiKit with NumPy 2.0? It should solve this issue for good. (Of course I don't know how much work it would entail....)

yaqwsx, 2024-07-23:
There is no compilation of KiKit. The error message suggests a corrupted Python environment that is installed by KiCAD. This should be KiKit independent, KiKit just triggers this behavior as it tries to use NumPy.

I believe the uninstallation didn't clean up all the Python files it installed.

viktak, 2024-07-23:
Ah, I see. Any idea where those files might be? After removing KiCAD, the installation folder is gone, so they may be somewhere else..

yaqwsx, 2024-07-23:
You can try running in KiCAD Command prompt: `pip install --upgrade --force-reinstall kikit`. That should force pip to revisit all versions of packages and hopefully bring them into a consistent versions.

viktak, 2024-07-23:
hah! This worked!!!! Here is the proof:
```bash
D:\Documents\KiCAD\8.0>pip install --upgrade --force-reinstall kikit
Defaulting to user installation because normal site-packages is not writeable
Collecting kikit
  Using cached KiKit-1.6.0-py3-none-any.whl.metadata (5.7 kB)
Collecting click>=7.1 (from kikit)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting commentjson>=0.9 (from kikit)
  Using cached commentjson-0.9.0-py3-none-any.whl
Collecting markdown2>=2.4 (from kikit)
  Downloading markdown2-2.5.0-py2.py3-none-any.whl.metadata (2.2 kB)
Collecting numpy (from kikit)
  Downloading numpy-2.0.1-cp311-cp311-win_amd64.whl.metadata (60 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.9/60.9 kB 3.2 MB/s eta 0:00:00
Collecting pcbnewTransition<=0.5,>=0.4.1 (from kikit)
  Using cached pcbnewTransition-0.4.1-py3-none-any.whl.metadata (1.1 kB)
Collecting pybars3>=0.9 (from kikit)
  Using cached pybars3-0.9.7-py3-none-any.whl
Collecting shapely>=2.0.3 (from kikit)
  Downloading shapely-2.0.5-cp311-cp311-win_amd64.whl.metadata (7.2 kB)
Collecting solidpython>=1.1.2 (from kikit)
  Using cached solidpython-1.1.3-py3-none-any.whl.metadata (1.4 kB)
Collecting colorama (from click>=7.1->kikit)
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting lark-parser<0.8.0,>=0.7.1 (from commentjson>=0.9->kikit)
  Using cached lark_parser-0.7.8-py2.py3-none-any.whl
Collecting PyMeta3>=0.5.1 (from pybars3>=0.9->kikit)
  Using cached PyMeta3-0.5.1-py3-none-any.whl
Collecting PrettyTable==0.7.2 (from solidpython>=1.1.2->kikit)
  Using cached prettytable-0.7.2-py3-none-any.whl
Collecting euclid3<0.2.0,>=0.1.0 (from solidpython>=1.1.2->kikit)
  Using cached euclid3-0.1-py3-none-any.whl
Collecting ply<4.0,>=3.11 (from solidpython>=1.1.2->kikit)
  Downloading ply-3.11-py2.py3-none-any.whl.metadata (844 bytes)
Collecting pypng<0.0.20,>=0.0.19 (from solidpython>=1.1.2->kikit)
  Using cached pypng-0.0.19-py3-none-any.whl
Using cached KiKit-1.6.0-py3-none-any.whl (252 kB)
Using cached click-8.1.7-py3-none-any.whl (97 kB)
Downloading markdown2-2.5.0-py2.py3-none-any.whl (47 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 47.2/47.2 kB 2.3 MB/s eta 0:00:00
Using cached pcbnewTransition-0.4.1-py3-none-any.whl (6.9 kB)
Downloading shapely-2.0.5-cp311-cp311-win_amd64.whl (1.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 9.2 MB/s eta 0:00:00
Downloading numpy-2.0.1-cp311-cp311-win_amd64.whl (16.6 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.6/16.6 MB 11.1 MB/s eta 0:00:00
Using cached solidpython-1.1.3-py3-none-any.whl (2.7 MB)
Using cached ply-3.11-py2.py3-none-any.whl (49 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Installing collected packages: pypng, PyMeta3, PrettyTable, ply, lark-parser, euclid3, solidpython, pybars3, pcbnewTransition, numpy, markdown2, commentjson, colorama, shapely, click, kikit
  Attempting uninstall: pypng
    Found existing installation: pypng 0.0.19
    Uninstalling pypng-0.0.19:
      Successfully uninstalled pypng-0.0.19
  Attempting uninstall: PyMeta3
    Found existing installation: PyMeta3 0.5.1
    Uninstalling PyMeta3-0.5.1:
      Successfully uninstalled PyMeta3-0.5.1
  Attempting uninstall: PrettyTable
    Found existing installation: prettytable 0.7.2
    Uninstalling prettytable-0.7.2:
      Successfully uninstalled prettytable-0.7.2
  Attempting uninstall: ply
    Found existing installation: ply 3.11
    Uninstalling ply-3.11:
      Successfully uninstalled ply-3.11
  Attempting uninstall: lark-parser
    Found existing installation: lark-parser 0.7.8
    Uninstalling lark-parser-0.7.8:
      Successfully uninstalled lark-parser-0.7.8
[... 36 lines cut ...]
  Attempting uninstall: kikit
    Found existing installation: KiKit 1.6.0
    Uninstalling KiKit-1.6.0:
      Successfully uninstalled KiKit-1.6.0
Successfully installed PrettyTable-0.7.2 PyMeta3-0.5.1 click-8.1.7 colorama-0.4.6 commentjson-0.9.0 euclid3-0.1 kikit-1.6.0 lark-parser-0.7.8 markdown2-2.5.0 numpy-1.26.4 pcbnewTransition-0.4.1 ply-3.11 pybars3-0.9.7 pypng-0.0.19 shapely-2.0.5 solidpython-1.1.3

D:\Documents\KiCAD\8.0>kikit --help
Usage: kikit [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  drc       Validate design rules of the board
  export    Export KiCAD boards
  fab       Export complete manufacturing data for given fabrication houses
  modify    Modify board items
  panelize  Panelize boards
  present   Prepare board presentation
  separate  Separate a single board out of a multi-board design.
  stencil   Create solder paste stencils

D:\Documents\KiCAD\8.0>
```

closed 2024-07-23 by viktak

StecklerCircuits, 2024-10-03:
Had to do this for Version 8.0.5 today... 
`pip install --upgrade --force-reinstall kikit`
might want to put that somewhere on the  FAQ page.
thanks again for KiKit !!