yaqwsx/KiKit #721: KiKit CLI Crashes (compatibiltiy issues)

greenscreenflicker, 2024-07-29:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

latest (output cannot be posted due to crash)

### KiCAD version

8.0.4

### Operating system

Windows

### Description

When running the commend, i get the following error output:

===================

C:\Users\mchhe\OneDrive\Dokumente\git\zn0_bms_v2\pcb\v0.3\fab>kikit fab jlcpcb --no-drc --assembly --schematic ../zn0_bms.kicad_sch ../zn0_bms.kicad_pcb gerber_single/

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\Scripts\kikit.exe\__main__.py", line 4, in <module>
    from kikit.ui import cli
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 6, in <module>
    from kikit.common import normalize
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\common.py", line 13, in <module>
    from shapely.geometry import LinearRing
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\shapely\__init__.py", line 13, in <module>
    from shapely.lib import GEOSException  # NOQA
AttributeError: _ARRAY_API not found
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\Scripts\kikit.exe\__main__.py", line 4, in <module>
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 6, in <module>
    from kikit.common import normalize
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\kikit\common.py", line 13, in <module>
    from shapely.geometry import LinearRing
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\8.0\3rdparty\Python311\site-packages\shapely\__init__.py", line 13, in <module>
    from shapely.lib import GEOSException  # NOQA
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: numpy.core.multiarray failed to import

### Steps to Reproduce

Execute commend above!

yaqwsx, 2024-07-29:
This is a duplicate of #706, #715, #720, and #717.

There is a packaging problem in KiCAD 8.0.4 that can put Python environment into an inconsistent state. The solution is to run: `pip install --upgrade --force-reinstall kikit`

kaeptn-h, 2024-08-01:
Works perfect for me

closed 2024-08-01 by yaqwsx