yaqwsx/KiKit #723: kikit fab does not honor DRC exclusions

rsholmes, 2024-08-09:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

version 1.6.0

### KiCAD version

Version: 8.0.4-8.0.4-0~ubuntu22.04.1

### Operating system

Linux Mint 21.2

### Description

Attached PCB has several courtyard errors excluded in DRC, so it passes DRC. But these exclusions are not honored by kikit and so it fails (unless I use `--no-drc`)
[busboard10.kicad_pcb.zip](https://github.com/user-attachments/files/16565864/busboard10.kicad_pcb.zip)

### Steps to Reproduce

```
$ kikit fab jlcpcb Hardware/busboard10/busboard10.kicad_pcb foo
** Found 16 DRC violations: **
[courtyards_overlap]: Courtyards overlap
    Rule: board setup constraints courtyard; Severity: error
    @(154.0000 mm, 57.5000 mm): Footprint J7
    @(147.7500 mm, 54.0000 mm): Footprint REF**
...
```

yaqwsx, 2024-08-10:
PCB exclusions are saved in the project file, not the PCB file. You provided only the board file, hence, I can't reproduce the issue.

rsholmes, 2024-08-10:
Project directory: 
[busboard10.zip](https://github.com/user-attachments/files/16569282/busboard10.zip)

yaqwsx, 2024-08-10:
If I run KiKit fab or drc on the board you provided, it passes without errors. 

I believe that you marked the exclusions in KiCAD and run KiKit just after that. Unfortunately, KiCAD saves the exclusions to the project file only when KiCAD closes. Hence, KiKit reads from the disk project file that has no exclusions marked. Please, try it again, but first, close the whole **project** before running KiKit.

This is quite stupid behavior of KiCAD and I have no way of fixing this from KiKit.

rsholmes, 2024-08-10:
Right, that works. Thanks.

closed 2024-08-10 by rsholmes