yaqwsx/KiKit #724: An error occurred: argument 3: TypeError: wrong type when building tightframe

harm363, 2024-08-14:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

kikit, version 1.5.1-1

### KiCAD version

8.0.1 

### Operating system

debian sid (kicad8_auto docker from kibot)

### Description

When i try to build a panel from a project it fails when i try to build projects with a tightframe instead of a normal frame.

I do run everything via the kibot_auto docker and these projects did work with the kicad7 docker, so i am sure i missed a change in how to configure the tightframe. 
I see no difference in panelizing rectangular boards or other shapes, i get a panel with frame type: frame, but not with frame type: tightframe.

running with debug trace gives :
``` bash
An error occurred: argument 3: TypeError: wrong type
No output files produced
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 279, in doPanelization
    frameCuts = ki.buildFraming(preset, panel)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui_impl.py", line 477, in buildFraming
    panel.makeTightFrame(framingPreset["width"], framingPreset["slotwidth"],
  File "/usr/lib/python3/dist-packages/kikit/panelize.py", line 1430, in makeTightFrame
    boardSlot = boardSlot.buffer(slotwidth, join_style="mitre")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/shapely/geometry/base.py", line 632, in buffer
    self._lgeos.GEOSBufferParams_setJoinStyle(params, join_style)
ctypes.ArgumentError: argument 3: TypeError: wrong type
make: *** [.CI/Makefile:47: run_prefligth] Error 1
```

I am probably missing something simple or I did something stupid but i dont know where to continue searching from this debug output.

If you have any questions do not hesitate to ask, i will do my best to answer quickly. 

### Steps to Reproduce

the command i use:
```bash 
	kikit panelize \
		-p "$CI_CONFIG"/kikit.json \
		-p "$USER_CONFIG"/project_kikit.json \
		"$SOURCE"/"$PNAME".kicad_pcb \
		"$FAB"/"$PNAME"_panel.kicad_pcb; 
```

my kikit.json config: 
```json
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 1,
        "hspace": "10mm",
        "vspace": "10mm",
        "rotation": "0deg",
        "alternation": "none",
        "renamenet": "Board_{n}-{orig}",
        "renameref": "{orig}"
    },
    "source": {
        "type": "auto",
        "tolerance": "25mm"
    },
    "tabs": {
        "type": "fixed",
        "width": "4mm",
        "vcount": 2,
        "hcount": 2,
        "source": "none",
        "tabfootprints":"_EVAbits_Panelizer:tab"    
    },
    "cuts": {
        "type": "mousebites",
        "prolong": "0.5mm"
    },
    "framing": {
        "type": "tightframe",
        "slotwidth": "2mm",
        "width": "5mm",
        "cuts": "h",
        "fillet": "1mm",
        "mintotalwidth": "70mm",
        "mintotalheight": "66mm"
    },   
    "post": {
        "type": "auto",
        "millradius": "0mm",
        "copperfill": false,
    },
}
```

my project_kikit.json config
```json
{
    "layout": {
        "rows": 1,
        "cols": 1,
        "hspace": "10mm",
        "vspace": "10mm",
        "rotation": "90deg"
    },
    "tabs": {
        "vcount": 2,
        "hcount": 2
    },
    "debug": {
        "trace": true
    }

}
```

yaqwsx, 2024-08-15:
I believe this was already fixed in the version 1.6.0 (as it works when I test it). Could you confirm that you don't observe the problem with the newest version?

harm363, 2024-08-15:
hmm, when I build the panel locally with version 1.6.0 it works. it seems the problem is somewhere with kikit in the kibot docker image. i wil contact its maintainer. in any case many thanks for your reply,

harm363, 2024-08-15:
yep, when i manually install kikit in the kibot docker my project is panelized correctly. very strange

set-soft, 2024-08-16:
> ### KiKit version
> kikit, version 1.5.1-1
> 
> ### KiCAD version
> 8.0.1

Get the *latest* images, they have KiCad 8.0.3 and KiKit 1.6.0 (KiBot 1.7.0)

harm363, 2024-08-16:
i have, i did pull the latest kicad 8 docker and installed the latest kikit version. that seems to work.

my current dockerfile:

```docker
FROM ghcr.io/inti-cmnb/kicad8_auto:latest

RUN  sed -i -e's/ main/ main contrib non-free/g' /etc/apt/sources.list  && \
	apt-get update  && \ 
    apt install -y jq pip zip python3-pandas python3-semver 
RUN apt remove -y kikit
    
#RUN dl_deb.py INTI-CMNB/KiKit --skip kikit-doc 
#RUN apt -y install --no-install-recommends ./*.deb && \
#    apt-get -y autoremove && \
#    rm /*.deb && \
#    rm -rf /var/lib/apt/lists/* /var/cache/debconf/templates.dat-old /var/lib/dpkg/status-old

RUN pip install --break-system-packages kikit

COPY entry.sh /entry.sh
RUN chmod a+x /entry.sh

WORKDIR /mnt

ENTRYPOINT [ "/entry.sh" ]
```

set-soft, 2024-08-16:
> i have, i did pull the latest kicad 8 docker and installed the latest kikit version via pip. that seems to work.

Current docker images already contains KiKit 1.6.0, no need to install it.

harm363, 2024-08-16:
> BTW: I don't recommend installing KiKit using pip in KiBot images, I provide Debian packages that integrate with the system (pip is 100% alien, this is why you need --break-system-packages)
I know, the pip install was purely to test if v1.6.0 worked. 

I already pull the latest kicad8_auto image
``` FROM ghcr.io/inti-cmnb/kicad8_auto:latest ```
that gave me crashes, but i will test again.
When i get home tonight i will rebuild my docker with: 
``` docker 
FROM ghcr.io/inti-cmnb/kicad8_auto:latest

RUN  sed -i -e's/ main/ main contrib non-free/g' /etc/apt/sources.list  && \
	apt-get update  && \ 
    apt install -y jq pip zip python3-pandas python3-semver 

COPY entry.sh /entry.sh
RUN chmod a+x /entry.sh

WORKDIR /mnt

ENTRYPOINT [ "/entry.sh" ]
```
That should get all the correct versions rigth?

set-soft, 2024-08-16:
> I already pull the latest kicad8_auto image
> `FROM ghcr.io/inti-cmnb/kicad8_auto:latest`

Yes, this contains KiKit 1.6.0, but remember to pull a fresh image, if you get 1.5.1 is because you are using a cached copy.

harm363, 2024-08-17:
well, i dont really know what happened. it works now. 
I had multiple kicad projects in the same folder for some reason, maybe that was the reason. anyway many thanks your both of your answers it really helped.

closed 2024-09-29 by yaqwsx