yaqwsx/KiKit #731: "Transform It" plugin doesn't work when KiKit is installed

Morad-T, 2024-09-09:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 10 19045.4780

### Description

This is already a known bug that has supposedly been patched before but I guess keeps returning?

https://github.com/openscopeproject/TransformIt/issues/1
https://github.com/yaqwsx/pcbnewTransition/issues/3

### Steps to Reproduce

Install KiKit and Transform It
Transform It stops performing it's functions
Removing KiKit backend resolves the issue

yaqwsx, 2024-09-10:
Can you please share the version of the pcbnewTransition?

yaqwsx, 2024-09-11:
E.g., by invoking `python -c "import pcbnewTransition; print(pcbnewTransition.__version__)"` in the KiCAD command prompt.

Morad-T, 2024-09-11:
I just ran that command
output says it's version 0.4.2

[image]

Morad-T, 2024-09-14:
Is there anything else required on my part?
also is there a workaround other than uninstalling and reinstalling the kikit backend?

yaqwsx, 2024-09-15:
I am not able to reproduce the issue you report. I just installed the transform it plugin and it works just fine side-by-side with KiKit installed. 

Could you open Preferences -> Action Plugin and report any error logs from plugin you can find there?

Morad-T, 2024-09-15:
Somehow it now works?
I'm really not sure what has changed since I was reporting the issue but I'm guessing that as I used to have Kikit installed previously a long time ago the dependency packages didn't get flushed.

Thank you for effort

closed 2024-09-15 by Morad-T