yaqwsx/KiKit #736: kikit 1.6.0 doesn't work with kicad 8.0.4 and 8.0.5, needs numPy 2.0 instaed of 1.26.4

M-oster, 2024-09-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

A module that was compiled using NumPy 1.x cannot be run in NumPy 2.0.1 as it may crash. To support both 1.x and 2.x versions of NumPy, modules must be compiled with NumPy 2.0. Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

### KiCAD version

8.0.5

### Operating system

Windows 11

### Description

last functioning kicad version was 8.0.3, with the later versions kikat doesn't work because of older numPy version 1.26.4, needs v2.0

### Steps to Reproduce

install of kicad 8.0.4 or 8.0.5. Uninstall kikat and reinstall doesn't fix the issue.

yaqwsx, 2024-09-17:
This is a duplicate of #720. Please follow the instructions in https://github.com/yaqwsx/KiKit/issues/720#issuecomment-2245595775.

PS: The project is called KiKit, not KiKat.

M-oster, 2024-09-17:
many thanks for the hint, haven't seen ist.
"pip install --upgrade --force-reinstall kikit" fixes the issue

closed 2024-09-17 by M-oster