yaqwsx/KiKit #737: error running separate command with rectangle source

wendallsan, 2024-09-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 11

### Description

I've designed a PCB per the instructions at [Multi-board workflow with KiKit](https://github.com/yaqwsx/KiKit/blob/master/docs/multiboard.md) and measured the top left and bottom right coordinates in the KiCAD PCB editor for the 3 boards I hope to derive.  When I enter the command according to the documentation, I get an error.  

Here is the command I'm executing in the KiCAD 8.0 Command Prompt, from within the directory where my kicad_pcb file is located:

```
kikit separate --source 'rectangle; tlx: 23mm; tly: 27mm; brx: 40mm; bry: 138mm' euro-mini-mixer-v2.kicad_pcb euro-mini-mixer-v2-bottom.kicad_pcb
```
Here is the error message I'm receiving:  
```
Error: Got unexpected extra arguments (tly: 27mm; brx: 40mm; bry: 138mm' euro-mini-mixer-v2.kicad_pcb euro-mini-mixer-v2-bottom.kicad_pcb)
```
[image]

Is there something I'm doing wrong?  Thank you for any assistance.

### Steps to Reproduce

Run command in the KiCAD 8.0 Command Prompt, from within the directory where the source kicad_pcb file is located:

```
kikit separate --source 'rectangle; tlx: 23mm; tly: 27mm; brx: 40mm; bry: 138mm' euro-mini-mixer-v2.kicad_pcb euro-mini-mixer-v2-bottom.kicad_pcb
```

yaqwsx, 2024-09-18:
You have to use double quotes (i.e., `"` instead of `'`) on Windows.

wendallsan, 2024-09-19:
That worked!  Thank you.

closed 2024-09-19 by wendallsan