yaqwsx/KiKit #738: kikit increases size of Silkscreen

dubstepenjoyer, 2024-09-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

8.0.4

### Operating system

Ubuntu 22.04

### Description

Hi. Sorry by using old version, but i dont know  what to do. 
In the project the Silkscreen layer looks like the picture below.
[image]
But after I start using the kikit, what happens is shown in the picture below.
[image]
This has never happened before, maybe it is connected with the transition to the version of Kikad 8 from Kikad 7. 
I think this is due to the fact that in the arguments in the docker file for the kikit image I have the version of kicad 7 specified, but I work with v8.
[Dockerfile](https://github.com/bobrovvv/KiKit/blob/master/Dockerfile)
this link on dockerfile that i describe above.

### Steps to Reproduce

I have yaml file that contains kikit script, after kikit give me file, i can see this wrong size of Silkscreen text.

yaqwsx, 2024-09-18:
You haven't provided enough information to reproduce the issue. Hence, I am afraid I can't help.

dubstepenjoyer, 2024-09-18:
What information needs to be provided?

yaqwsx, 2024-09-18:
A step-by-step guide with all the input data so the issue can be reproduced.

To be honest, I don't even fully understand where the issue is. You showed a screenshot from Pcbnew and then a render from PcbDraw. Yes, the silkscreen is larger. **But how was the board transformed?** Why are the pictures from different tools and not from the same one?

yaqwsx, 2025-01-26:
No further information was provided, closing. Feel free to reopen if you supply enough information to reproduce and trace the issue.

closed 2025-01-26 by yaqwsx