yaqwsx/KiKit #744: Kikit space between tab and board edge and also issue in mousebits

bavanz, 2024-10-01:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 11

### Description

https://forum.kicad.info/t/kikit-panel-issue-on-edge-cut-and-mousebits/55136?u=bavanz

### Steps to Reproduce

https://forum.kicad.info/t/kikit-panel-issue-on-edge-cut-and-mousebits/55136?u=bavanz

yaqwsx, 2024-10-01:
I am sorry, the steps for reproductions are incomplete - there are no files to run the commands on. There won't be any action taken until they are completed.

yaqwsx, 2024-11-12:
As there is no reproducer, closing.

closed 2024-11-12 by yaqwsx