yaqwsx/KiKit #747: No board edges found

bobatsar, 2024-10-10:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.5

### Operating system

Ubuntu 24.04

### Description

As already mentioned in #352 it seems this bug reappeared.

### Steps to Reproduce

```
kikit panelize \
    --layout 'grid; rows: 2; cols: 2;' \
    --tabs full \
    --cuts vcuts \
    test.kicad_pcb panel.kicad_pcb
An error occurred: No board edges found
No output files produced
```

[kikit_no_board_edges_found.zip](https://github.com/user-attachments/files/17328037/kikit_no_board_edges_found.zip)

When 
```
		(44 "Edge.Cuts" user "Center.EdgeCuts")
```
is changed to 

```
		(44 "Edge.Cuts" user)
```

Another error occurs (other ticket will be opened)

yaqwsx, 2024-10-11:
Fixed in 309b937.

closed 2024-10-11 by yaqwsx

bobatsar, 2024-10-11:
Wow, that was fast. 
Tested in `kikit, version 1.6.0+7.g309b937` and works.

Thanks