yaqwsx/KiKit #748: Multiple outlines

bobatsar, 2024-10-10:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.5

### Operating system

Ubuntu 24.04

### Description

I have a project with a complicated outline.
When trying to panelize it, I get the error 

```
An error occurred: test.kicad_pcb: Multiple outlines (4) at [144.694624, 95.943444]
No output files produced
```

I can't find the error in my files as at the specified coordinate is no outline at all.

- When the inner outline is removed, kikit runs without error but the result does not reflect the original outline at all.
- When removing the outer outline kikit runs without error but also the result does not reflect the original.

### Steps to Reproduce

Run 

```
❯ kikit panelize \
    --layout 'grid; rows: 2; cols: 2;' \
    --tabs full \
    --cuts vcuts \
    test.kicad_pcb panel.kicad_pcb
An error occurred: test.kicad_pcb: Multiple outlines (4) at [144.694624, 95.943444]
```
with the example files.

[kikit_multiple_outlines.zip](https://github.com/user-attachments/files/17328135/kikit_multiple_outlines.zip)

yaqwsx, 2024-10-11:
This was fixed in 46b2ea9.

The problem is that KiCAD's polygons don't use properties `GetStart()` and `GetEnd()` for their start/end points. They always return a fixed coordinate. However, we have never noticed as all our test cases used a single polygon (possibly with holes)

closed 2024-10-11 by yaqwsx

bobatsar, 2024-10-11:
Thanks for the fix. 

The error is now gone but the outline in the panel does not match the original at all. I just opened #749.

tested with `kikit, version 1.6.0+7.g309b937`