yaqwsx/KiKit #749: Malformed outline on panel

bobatsar, 2024-10-11:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0+7.g309b937

### KiCAD version

8.0.5

### Operating system

Ubuntu 24.04

### Description

A outline imported from dxf files is unfortunately malformed in the resulting panel.

test_outline:
[image]

The input outlines render normally in the PCB-3D viewer (correct closed outline for KiCad).
[image]

but the resulting panel is messed up.
[image]

### Steps to Reproduce

Use the example pcbs in

[kikit_malformed_outline.zip](https://github.com/user-attachments/files/17340292/kikit_malformed_outline.zip)

and run e.g.
``` 
kikit panelize \
    --layout 'grid; rows: 2; cols: 2;' \
    --tabs full \
    --cuts vcuts \
    test_outline.kicad_pcb panel_outline.kicad_pcb
```

bobatsar, 2024-10-21:
@yaqwsx is there more input required to handle this case? Can I provide more data for debugging or can you help me debug this myself?

It would be very helpful for us to fix this bug.

yaqwsx, 2024-11-12:
I can confirm it is the same issue as #750. It is a bug in the full tabs generator. Similarly to the author of #750, I believe you don't don't to actually use the full tabs (as they would completely cover the shape of your PCB). Use fixed or annotation tabs instead.

greenscreenflicker, 2024-11-12:
it seems that the full tab generator works when the tab side is a plain line. 
If you can't find a clever solution with the script (which is obviously to prefer), it would be good to output a warning message.