yaqwsx/KiKit #751: An error occurred: 'LineString' object is not subscriptable

greenscreenflicker, 2024-10-16:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.5

### Operating system

Windows 11

### Description

When a too long hwidth is chosen the following error comes up:

C:\Users\mchhe\OneDrive\Dokumente\git\dinrail_ac_pd\pcb\acpd_v1\panel>kikit panelize  --layout "grid; rows: 1; cols: 5; space: 4mm; rotation: 0deg;"  --tabs "fixed; hwidth: 46mm; vwidth: 60mm" --cuts "vcuts" --post "millradius: 0.3mm"   --s
ource "tolerance: 20mm"  ../acpd_v1.kicad_pcb panel.kicad_pcb
An error occurred: 'LineString' object is not subscriptable
No output files produced

### Steps to Reproduce

Use the PCB recently send and observe the error.above

BogdanTheGeek, 2024-11-07:
I have had the same issue, tracked it down to my board not being symmetrical. what happens if you alternate the rows so the matching sides are identical?

yaqwsx, 2024-11-12:
Fixed in d3d118f7b9c98ed544bcf8cb898487bb42216159.

The problem was in error reporting when the users tries to cut a curve using a V-cut.

closed 2024-11-12 by yaqwsx