yaqwsx/KiKit #754: KiKit places vcuts at wrong position

greenscreenflicker, 2024-10-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit --version

### KiCAD version

8.0.5

### Operating system

Windows 11

### Description

Kikit places vcuts at obviously wrong position.

[image]

### Steps to Reproduce

Use panalizing command with PCB Send via email and observe same error

kikit panelize  --layout "grid; rows: 1; cols: 5; space: 20mm; rotation: 90deg;"  --tabs "fixed; hwidth: 10mm; vwidth: 60mm" --cuts "vcuts" --post "millradius: 0.3mm"   --source "tolerance: 20mm"  ../acpd_v1.kicad_pcb panel.kicad_pcb

yaqwsx, 2024-10-19:
Well, where should they be placed, in your opinion? Please draw a picture.

I think the V-cuts are placed in the right position - where your tab meets the original PCB edge. I would say that your positioning of tabs in combination with V-cuts doesn't make sense.

yaqwsx, 2024-11-12:
If the V-cuts were placed as you suggest, then the shape of PCB change as well. As I already said, I think the problem is in the position of you tabs. Please, have a look at the panel in the 3D Viewer - you will see the placement of the tabs makes no sense in this context. Please, place the tabs via manual annotation and everything will be as you expect.

yaqwsx, 2024-11-12:
Based on your experience, I implemented a check if a V-cut damages the original PCBs in f972993dfdda8c17ce18ecde25d674b7c9391dad. The error message should clearly say what is the problem and suggest actions the user can take.

[image]