yaqwsx/KiKit #755: annotated tabs not generated with small pcb

Nightreaver, 2024-10-20:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.4

### Operating system

Windows 10

### Description

I been trying to make a small round shaped pcb and I put a single tab footprint on one side.

[image]

Then I tried to use this tab with several rails, and most of them failed to generate

Tightframe:
[image]

Frame:
[image]

Railstb (at least seem to connect but missing mousebites):
[image]

I dont get any error messages tho.

generated the board with couple of setting through json `kikit panelize -p panel-config.json motor_pcb.kicad_pcb panel.kicad_pcb`

```
{
    // There can be C-like comments
    "layout": {
        "type": "grid",
        "rows": 2,
        "cols": 4,
        "hspace": "2mm",
        "vspace": "2mm",
        "rotation": "180deg",
        "alternation": "rows",
        "renamenet": "Board_{n}-{orig}",
        "renameref": "{orig}",
    },
    "tabs": {
        "type": "annotation",
    },
    "cuts": {
        "type": "mousebites",
        "drill": "0.5mm",
        "spacing": "0.8mm",
        "prolong": "0.5mm",
        "offset": "-0.2mm"
    },
    "framing": {
        "type": "frame",
        "width": "3mm",
        "space": "2mm",
    },
    "text": {
      "type": "simple",
      "text": "JLCJLCJLCJLCJLCJLCJLCJLC",
      "anchor": "mb",
      "voffset": "-1.5mm",
      "hjustify": "center",
      "vjustify": "center",
      "height": "1.5mm",
      "layer": "F_SilkS",
    },
    "post": {
        "type": "auto",
        "millradius": "0mm",
        "copperfill": false
    }
}
```

### Steps to Reproduce

Use this pcb to generate the panel [motor_pcb.zip](https://github.com/user-attachments/files/17451826/motor_pcb.zip)

yaqwsx, 2024-10-20:
Your tab symbol is called `Library:Tab`. However, KiKit recognizes only `KiKit:Tab` as a tab symbol. You are not using the KiKit's library. Hence, no tabs are rendered.

If you have a good reason for not using the official library, you can specify tab footprint name via `tabfootprints: Library:Tab` in the tabs section. With this change or using the KiKit's library you get the expected result:

[image]
[image]

closed 2024-10-20 by yaqwsx

Nightreaver, 2024-10-20:
Thats an easy fix! thanks a lot