yaqwsx/KiKit #756: Error 'PCB_TEXT' object has no attribute 'GetWidth'

dartrax, 2024-10-21:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.6

### Operating system

Windows 10

### Description

When I want to panelize again (i.e. after checking the first panelization I want to change a parameter in the Panelize GUI), I'm getting the error message:
"Cannot perform: 'PCB_TEXT' object has no attribute 'GetWidth'"

### Steps to Reproduce

The command is:
```
kikit panelize^
    --layout "rows: 5" ^
    --tabs "type: full" ^
    --cuts "type: vcuts; linewidth: 0.1mm; layer: Edge.Cuts" ^
    "D:\Projekte\...v3.2.kicad_pcb" "D:\Projekte\...v3.2_panelized.kicad_pcb"
```

yaqwsx, 2024-10-21:
There is not enough information provided to reproduce this issue (no input files).

dartrax, 2024-10-21:
I observed this first when I set the Cuts layer to Edge.Cuts.
May be this helps, if not I'll try to reproduce that with some example file later.

dartrax, 2024-10-24:
I just tested again with this Testfile and I'm getting the same error:
https://forum.kicad.info/t/is-there-a-good-resource-with-example-kicad-pcb-files/30378/3

Please let me know when you need further information to reproduce it.

yaqwsx, 2024-10-24:
Thank you for the example. I just checked with the upstream version, and it seems to be working just fine. Could you upgrade to the upstream version and report if you still perceive the problem?

dartrax, 2024-10-24:
I upgradet to the upstream version using the doc here https://yaqwsx.github.io/KiKit/latest/installation/upgrading/ 
`pip install https://github.com/yaqwsx/KiKit/archive/master.zip`
Was that the right way, do I need to upgrade the GUI, too? Because I'm still getting the error message.

yaqwsx, 2024-10-24:
Then I guess this will be platform-related (I tested it on Linux).

yaqwsx, 2024-10-29:
I just tested the issue with the board you provided on:
- Ubuntu 22.04 with KiCAD 8.0.6,
- Ubuntu 24.04 with KiCAD 8.0.6,
- Windows 11, KiCAD 8.0.2 and 8.0.6 
[image]

I wasn't able to reproduce the issue. I don't suspect a serious difference between Windows 10 and 11 in this regard.

I encourage you to ensure you are running the latest version of KiKit with all the dependencies via  `pip install --upgrade --force-reinstall kikit`. Then, provide **precise steps for reproduction**. If you fail to provide them within 10 days, I will consider this issue resolved.

f1gwr, 2024-10-31:
I'm getting this message too in Linux Mint MATE 21.3 with Kicad 8.0.6 and kikit 1.6 which were just force-reinstalled. And defining V-cuts in Edge.Cuts is mandatory at JLCPCB.
[PanelJson.json](https://github.com/user-attachments/files/17590320/PanelJson.json)

yaqwsx, 2024-10-31:
I just tried to use your panel setting on one of the test files in Kikit suite with both, upstream and KiKit 1.6. Both panelized just fine:

[image]

KiCAD 8.0.6 and Ubuntu 22.04 were used.

There has to be something missing.

dartrax, 2024-11-03:
> I encourage you to ensure you are running the latest version of KiKit with all the dependencies via `pip install --upgrade --force-reinstall kikit`. Then, provide **precise steps for reproduction**. If you fail to provide them within 10 days, I will consider this issue resolved.

Hi, thanks for your ongoing effort understanding this issue.
I've closed Kicad, opened the Kicad Command prompt and did as you said:

[image]
...
[image]

1. Then I started Kicad 
2. created a new project in a subfolder of the project I want to panelize
3. opened the .kicad_pcb file
4. opened the GUI
5. imported the .json
6. set Input and Output file
7. Click "Panelize"
8. The Error message pops up
9. Click "Ok" to dismiss the Error message
10. Kicad crashes to desktop.

[image]

dartrax, 2024-11-03:
I made another test with =layer: F.SilkS=
I do not get the error message, instead Kicad directly crashes to deskop.
I think this are actually two issues. The Error message 'PCB_TEXT' is realted to the parameter =layer: Edge.Cuts=.
The crashing may be related to the directory structure, I'm currently testing that...

yaqwsx, 2024-11-04:
You failed to provide **precise** steps to reproduce as you didn't provide the source file where this happens. As stated in the issue template, you can share the files confidentially. Neither did you include the output with `--debug 'trace: true'`.

I just wasted a couple of hours hunting this down. If you had provided the source from the beginning or at least attached the debug trace (as instructed in the issue template), the bug could have been traced down in 10 minutes... 

The problem is that your **source** board contains a text on the `Edge.Cuts` layer and KiKit tries to interpret it as a PCB edge, but fails.  Fixed in f15d085653117cef8a5b69a4fc9a3e5443f329a1.

closed 2024-11-04 by yaqwsx

dartrax, 2024-11-04:
@yaqwsx again, thanks a lot for your effort and finding and fixing it!

Don't be so hard on yourself about the wasted time - if it had been that easy, you would have recreated the error with the source files I posted in [https://github.com/yaqwsx/KiKit/issues/756#issuecomment-2434724376]. Also, in the same post, I asked you to "let me know when you need further information to reproduce it." If it would have been as easy as `--debug 'trace: true'`, I'm sure you'd just asked for it. But solving bugs in software is not always a straightforward as it seems...