yaqwsx/KiKit #759: 3 bugs report: Kikit 1.6 crashes Kicad 8.0.6 Linux with high count panels (6x10 PCB's) + .json file amnesia + auto file extension missing

f1gwr, 2024-10-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6

### KiCAD version

8.0.6

### Operating system

Linux Mint MATE 21.3 up to date

### Description

1) Now my generated panel (h=6 v=10 PCB's over a 100x100mm panel) makes **Kicad crash** (always), and without any message. Short before with the same configuration I generated same panel size but with lower count PCB's (5x4) without crash (but still with json files amnesia - see below).
2) In addition the .json file **does not keep track** of the **input file** (sometimes) and **output file** (always) in kikit when loading former .json file
3) And oh when saving a json file, kikit could add a .json extension, otherwise later clicking the "load .json" button doesn't even see the file.
I'll send you the files by PM.

### Steps to Reproduce

Application: KiCad x86_64 on x86_64

Version: 8.0.6, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 9.0.0
	FontConfig 2.15.0
	libcurl/8.10.1 OpenSSL/3.3.2 zlib/1.3.1 libidn2/2.3.7 libpsl/0.21.5 nghttp2/1.63.0

Platform: Freedesktop SDK 24.08 (Flatpak runtime), 64 bit, Little endian, wxGTK, X11, mate, x11
OpenGL: NVIDIA Corporation, NVIDIA GeForce RTX 4070 Ti SUPER/PCIe/SSE2, 4.6.0 NVIDIA 550.107.02

Build Info:
	Date: Oct 14 2024 22:25:13
	wxWidgets: 3.2.6 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1
	ngspice: 43
	Compiler: GCC 14.2.0 with C++ ABI 1019

---------------------------------------
**.json file:**
{
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "rows": "10",
        "cols": "6"
    },
    "tabs": {
        "type": "fixed"
    },
    "cuts": {
        "type": "mousebites",
        "offset": "-0.25mm",
        "prolong": "0.6mm"
    },
    "framing": {
        "type": "railstb",
        "width": "6mm",
        "fillet": "2mm"
    },
    "tooling": {
        "type": "3hole",
        "hoffset": "4mm",
        "voffset": "3mm"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "8mm",
        "voffset": "3mm"
    },
    "text": {
        "type": "simple",
        "voffset": "3mm",
        "text": "Tek relais de remplacement - types 34/54/55"
    },
    "text3": {
        "type": "simple",
        "voffset": "-3mm",
        "text": "F1GWR - v1.0 oct 2024",
        "anchor": "mb"
    },
    "page": {
        "type": "D"
    },
    "post": {
        "copperfill": "True",
        "millradius": "1mm"
    }
}

f1gwr, 2024-10-29:
Ended up with more weird behaviour: the panelization only produced the panel cutout ; other layers were simply non-existent.
Finally after investigating, deleting the .config file, reinstalling everything and moreover... sleeping twice :)
now found out the most probable reason for that so called "bug": I've been using "v2.2" in the file names which is stupid as the dot means a filename extension, therefore messing up kikit!
After the dots suppression all seems working fine now.
Sorry for cluttering the forum with this oddity!
Maybe this could help someone making the same mistake in the future...

closed 2024-10-29 by f1gwr

yaqwsx, 2024-10-29:
KiKit shouldn't crash on names with dots inside. I will look into this. Thanks for your insight!

FractalEngineer, 2024-11-09:
Same issue here on Windows 11; kicad 8.0.6, Kikit 1.6 and master. Crashes on panelization

yaqwsx, 2024-11-10:
Unless you provide **precise** steps for reproduction (that is the source file, exact file names, settings) this won't be examined further.

f1gwr, 2024-11-11:
Files:
[KikitIssue759.zip](https://github.com/user-attachments/files/17708092/KikitIssue759.zip)

Steps:
Create a new folder, named Panel, inside the original (single) PCB folder TekRelays35_50_pcb
Create a new project named TekRelays35_50_panel stored in Panel folder
Open PCB editor in the project
Click Kikit icon that open the setting window
Input file: TekRelays35_50_pcb.kicad_pcb (attention: TekRelays35_50_**_pcb_**.kicad_pcb)
Output file TekRelays35_50_panel.kicad_pcb (attention: TekRelays35_50_**_panel_**.kicad_pcb)
settings are as in the enclosed json:
{
    "layout": {
        "hspace": "0.4mm",
        "vspace": "0.4mm",
        "rows": "11",
        "cols": "6"
    },
    "cuts": {
        "type": "vcuts",
        "linewidth": "0.4mm",
        "layer": "Edge.Cuts"
    }
}
(note: it is unclear when requesting vcuts if you have to specify the 0.4 mm V milling width in "layout/v&hspace" and/or "cuts/linewidth")
From there click the panelize button 
And Kicad crash!
If you open the resulting TekRelays35_50_panel.kicad_pcb, you will only get the panel outline, but not the 66 PCBs inside

yaqwsx, 2024-11-12:
I, unfortunately, struggle to reproduce your issue:

[image]

There are a couple of ideas what might be wrong:
- if you are working on a network drive, please, try it locally to see if the problem disappears (KiCAD is known to struggle with network locations)
- if you are working in area backed up by OneDrive, please, try it in a location outside OneDrive.

Also, you don't have to create a KiCAD project for the panel. That will happen automatically.

f1gwr, 2024-11-12:
I'm working only on a M2.SSD located on the computer motherboard since months with no issue.

When using the PCB editor instead of the full Kicad, i.e. without creating a new project, the said PCB editor doesn't crash. But I need to work from a copy of the initial TekRelays35_50_**_pcb_**.kicad_pcb because I get:
[image]
which is not acceptable (need to keep my original PCB file in case of further modifications), and also define an output file as: TekRelays35_50_**_panel_**.kicad_pcb
Suggestion: this could be done automatically by default.

Finally I succeeded after many trials and errors to find the correct settings:
**Command:**
kikit panelize \
    --layout 'hspace: 0.4mm; vspace: 0.4mm; rows: 11; cols: 6' \
    --tabs 'type: full' \
    --cuts 'type: vcuts; offset: -0.2mm; linewidth: 0.4mm; layer: Edge.Cuts' \
    '/home/cg/Documents/Kicad/Relais7A13/KiCAD/TekRelays35_50_pcb/TekRelays35_50_pcb.kicad_pcb' '/home/cg/Documents/Kicad/Relais7A13/KiCAD/TekRelays35_50_pcb/Panel/TekRelays35_50_panel.kicad_pcb'

**and settings:**
{
    "layout": {
        "hspace": "0.4mm",
        "vspace": "0.4mm",
        "rows": "11",
        "cols": "6"
    },
    "tabs": {
        "type": "full"
    },
    "cuts": {
        "type": "vcuts",
        "offset": "-0.2mm",
        "linewidth": "0.4mm",
        "layer": "Edge.Cuts"
    }
}

**NOW!
The remaining "bug" is:
 a sticky message comes out sometimes:
[image]
Could you please make clear what this is related to and how to avoid it definitely? As this has already pointed out by other users...**

yaqwsx, 2024-11-13:
To be honest, I need clarification on what you are saying. So, instead of responding, I will try to start with a blank canvas: The intended workflow with KiKit is:
- open a stand-alone instance of **Pcbnew**, not **KiCAD** (this is what the dialog is saying!)
- do penalization there.

Regarding the issue with `PCB_TEXT` - please use the upstream version. It should be already fixed.

Why do we require a stand-alone window to be opened? Well, KiCAD doesn't offer any API for opening/closing projects, so if people want GUI, we have to do it a window that the user already opened. Why a new window is needed? KiCAD crashes when you try to modify the currently opened file by an external tool.

f1gwr, 2024-11-15:
Thank you for your explanations which makes things clearer.
What should be made also clear would be:
- open a blank PCB editor or the single PCB?
- create manually a "panel" folder or will it be created automatically?
- is it necessary to duplicate the single PCB file with a copy inside the "panel" folder?
- what input file to choose in kikit settings: the original one or the copy in the "panel" folder?
- what is the use of the output file as the panelization results are finally stored in the input file?

yaqwsx, 2024-11-17:
Panelization takes a files and create a file. That's it. That should answer all your questions. Just to be sure let's cover your questions one-by-one

> open a blank PCB editor or the single PCB?

open a stand-alone instance of Pcbnew, not KiCAD (this is what the dialog is saying!)

> create manually a "panel" folder or will it be created automatically?

You specify the output file explicitly in the panelization dialog, why would you create anything before hang?

> is it necessary to duplicate the single PCB file with a copy inside the "panel" folder?
> what input file to choose in kikit settings: the original one or the copy in the "panel" folder?

The paneliazation process takes an input file. I don't understand your need to copy anything.

> what is the use of the output file as the panelization results are finally stored in the input file?

KiKit **never** touches your input files. Why would it modify your **inputs**?