yaqwsx/KiKit #761: Can't run Kikit at all with KiCad 8.0.6

Hyrla, 2024-10-29:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

Application: KiCad x64 on x64  Version: 8.0.6, release build  Libraries: 	wxWidgets 3.2.6 	FreeType 2.13.2 	HarfBuzz 9.0.0 	FontConfig 2.14.2 	libcurl/8.8.0-DEV Schannel zlib/1.3.1  Platform: Windows 11 (version 22631), édition 64-bit, 64 bit, Little endian, wxMSW  Build Info: 	Date: Oct 14 2024 01:02:33 	wxWidgets: 3.2.6 (wchar_t,wx containers) 	Boost: 1.85.0 	OCC: 7.8.1 	Curl: 8.8.0-DEV 	ngspice: 43 	Compiler: Visual C++ 1939 without C++ ABI  Build settings:

### Operating system

Windows 11

### Description

Hello, since I updated to KiCad 8.0, KiKit is very unstable. I can't run KiKit without crashing. Files are generated but KiCad has to be restarted each time. But since 8.0.6, KiKit can't run at all. I did try to reinstall plugin, pip library and KiCad it but it didn't work. Even with an "empty" panel (1x1 panel, default config for everything) it crashes.

[image]

### Steps to Reproduce

Here is the PCB I'm trying to panelize (very simple, 2 LEDs and one resistor)
[Elgo-lamp-white.zip](https://github.com/user-attachments/files/17556610/Elgo-lamp-white.zip)

yaqwsx, 2024-10-29:
This seems like broken dependencies (similar to #720) - could you reinstall all of them via `pip install --upgrade --force-reinstall kikit`?

Hyrla, 2024-10-29:
> This seems like broken dependencies (similar to #720) - could you reinstall all of them via `pip install --upgrade --force-reinstall kikit`?

Thank you for your reply. It did indeed fix this issue. But I am still facing [issue 694](https://github.com/yaqwsx/KiKit/issues/694). I close this one and continue talking on the other one.

[image]

closed 2024-10-29 by Hyrla