yaqwsx/KiKit #762: JLCPCB_CORRECTION wrong orientation on multiple components

greenscreenflicker, 2024-11-11:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.5

### Operating system

Windows 11

### Description

May it be that JLCPCB_CORRECTION is only applied to the first element of a series?
I see that the correction is applied for one, but not for the other. In the schematic both are there.

[image]

### Steps to Reproduce

Use PCB send. Apply
kikit panelize  --layout "grid; rows: 1; cols: 5; space: 4mm; rotation: 0deg;"  --tabs "fixed; hwidth: 42mm; vwidth: 60mm" --cuts "vcuts" --post "millradius: 0.3mm"   --source "tolerance: 20mm"  ../acpd_v2.kicad_pcb panel.kicad_pcb 
kikit fab jlcpcb --assembly  --schematic  ../acpd_v2.kicad_sch panel.kicad_pcb --no-drc gerber_panel/

Thank you for checking

yaqwsx, 2024-11-12:
You haven't provided all the necessary files for reproduction - acpd_v2.kicad_sch is missing.

yaqwsx, 2024-11-13:
I think your conclusion is incorrect. What happened is that JLC PCB changed the specification of the POS file. They seem to require the orientation of the components to be given when viewing from the corresponding side of the PCB. This didn't use to be the case.

I will need to contact JLC PCB and get a precise specification of what they require, so I can avoid making ad-hoc changes that can even make it worse.

greenscreenflicker, 2024-11-13:
@yaqwsx note that  both sot23 are on the same pcb side, so i don't think thats it

yaqwsx, 2024-11-14:
It is it. If you switch sides, the orientation of 90 degrees doesn't change, but the orientation of 0 degrees flips to 180 degrees.

greenscreenflicker, 2024-11-14:
Note that the green is correctly orientated while red is not. Both are the exact same component with the exact same jlcpcb correction parameter.

markh-de, 2024-11-17:
Yes, it would be great if you could contact JLCPCB and try to fix this issue (#664).  Even if you don't get an answer, we really know how to calculate the "correct" values, don't we?  If you want to avoid regression, the changed calculation feature could be a CLI option that users need to pass explicitly.

This issue just bit me.  I have some connectors that aren't rotated N×90°, and they are good on the top side, but have flipped angle interpretation on the bottom side.

Thanks for all your hard work.

markh-de, 2024-11-21:
I did the _180-angle_ conversion as suggested in #664, until I realized that this only works as long as you don't have orientation compensation != 0°.  That compensation data also needs to be flipped.  I hope that PR #767 helps fixing this issue.

markh-de, 2024-12-09:
@yaqwsx: Did you have a chance to test #767, yet? It looked very promising in my tests.

Any other users who are affected by this issue? Can you please try the patch and provide some feedback so that Jan can make a well-informed decision about the PR. Thank you!

yaqwsx, 2025-01-26:
This should be fixed in 8c980f3a245cd0d0f9a445c6c950817587129ed5.

closed 2025-01-26 by yaqwsx