yaqwsx/KiKit #763: Updated to V8.0.6, KiKit Numpy install error

NinoBenci, 2024-11-13:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

A module that was compiled using NumPy 1.x cannot be run in NumPy 2.1.2 as it may crash. To support both 1.x and 2.x versions of NumPy, modules must be compiled with NumPy 2.0. Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.  If you are a user of the module, the easiest solution will be to downgrade to 'numpy<2' or try to upgrade the affected module. We expect that some modules will need time to support NumPy 2.

### KiCAD version

8.0.6

### Operating system

Windows 10

### Description

Installed KiKit using KiCad 'Plugin an Content Manager'.
Then through the KiCad 8.0 Command Prompt issued the following command;

`C:\Program Files\KiCad\8.0>pip install kikit`

resulting in the following errr:

```
Installing collected packages: numpy
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '\\\\ad.monash.edu\\home\\User083\\ninob\\Documents\\KiCad\\8.0\\3rdparty\\Python311\\site-packages\\numpy\\core\\include\\numpy\\npy_interrupt.h'
Check the permissions.
```

The Win10 account has ADMIN rights

### Steps to Reproduce

Issuing the following;

`pip install kikit`

as described at: https://yaqwsx.github.io/KiKit/latest/installation/windows/#installation-on-windows

the installation fails.

yaqwsx, 2024-11-14:
This seems to be a duplicate of #736, #720 and #761. KiCAD 8.0.6 update a major dependency without updating transitive dependencies. The fix is to run `pip install kikit --force-reinstall`. Pip will freshly resolve the dependency tree and KiKit should work again.

closed 2024-11-14 by yaqwsx

NinoBenci, 2024-11-14:
Thank you for the update.