yaqwsx/KiKit #764: ufunc 'create_collection' not supported

christianhalter, 2024-11-14:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

Application: KiCad PCB Editor x64 on x64  Version: 8.0.6, release build  Libraries: 	wxWidgets 3.2.6 	FreeType 2.13.2 	HarfBuzz 9.0.0 	FontConfig 2.14.2 	libcurl/8.8.0-DEV Schannel zlib/1.3.1  Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW OpenGL: NVIDIA Corporation, NVIDIA GeForce RTX 3060/PCIe/SSE2, 4.6.0 NVIDIA 546.01  Build Info: 	Date: Oct 14 2024 01:02:33 	wxWidgets: 3.2.6 (wchar_t,wx containers) 	Boost: 1.85.0 	OCC: 7.8.1 	Curl: 8.8.0-DEV 	ngspice: 43 	Compiler: Visual C++ 1939 without C++ ABI  Build settings:

### Operating system

Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

### Description

When I try to panelize a board, this sign shows up and nothing is done:

[image]

### Steps to Reproduce

1. Open KiKit: Panelize PCB from PCB Editor 8.0 (standalone)
2. Import JSON configuration file (which worked some weeks ago)
3. Select input and output files
4. Click Panelize

yaqwsx, 2024-11-14:
This is a duplicate of https://github.com/yaqwsx/KiKit/issues/736, https://github.com/yaqwsx/KiKit/issues/720,  https://github.com/yaqwsx/KiKit/issues/761 and #764. KiCAD 8.0.6 update a major dependency without updating transitive dependencies. The fix is to run pip install kikit --force-reinstall. Pip will freshly resolve the dependency tree and KiKit should work again.

closed 2024-11-14 by yaqwsx

christianhalter, 2024-11-14:
Worked! Thanks, I didn't realize this was duplicated because I had a different message. Thanks again!