yaqwsx/KiKit #765: Cannot install KiKit on v8.06 under win 10

softfoot, 2024-11-16:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

Cannot install KiKit

### KiCAD version

Application: KiCad x64 on x64  Version: 8.0.6-82-ge7779d9b91, release build  Libraries: 	wxWidgets 3.2.6 	FreeType 2.13.3 	HarfBuzz 10.0.1 	FontConfig 2.15.0 	libcurl/8.10.1-DEV Schannel zlib/1.3.1  Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW  Build Info: 	Date: Nov 14 2024 04:22:13 	wxWidgets: 3.2.6 (wchar_t,wx containers) 	Boost: 1.86.0 	OCC: 7.8.1 	Curl: 8.10.1-DEV 	ngspice: 43 	Compiler: Visual C++ 1939 without C++ ABI  Build settings:

### Operating system

Windows 10 (uptodate)

### Description

I installed KiCAD v8.06 and installed kikit via PCM then when I tried to run the PCB editor I go the prompt to install kikit via the Kicad command prompt as usual.
I entered the command "pip install kikit" which appeared to be succesful and then tried to run "kikit --help" and got the following output

   File 
"C:\Users\Bob\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\common.py", 
line 13, in <module>
     from shapely.geometry import LinearRing
   File 
"C:\Users\Bob\Documents\KiCad\8.0\3rdparty\Python311\site-packages\shapely\__init__.py", 
line 13, in <module>
     from shapely.lib import GEOSException  # NOQA
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: numpy.core.multiarray failed to import

C:\Users\Bob\Documents\KiCad\8.0>

### Steps to Reproduce

Install KiCAD v8.06
Install KiKit via PCM
Start PCBNEW and get prompt to install "backend"
Install "Backend" per instructions on supplied link
Try to run KiKit --help

yaqwsx, 2024-11-17:
This is a duplicate of https://github.com/yaqwsx/KiKit/issues/736, https://github.com/yaqwsx/KiKit/issues/720, https://github.com/yaqwsx/KiKit/issues/761 and https://github.com/yaqwsx/KiKit/issues/764. KiCAD 8.0.6 update a major dependency without updating transitive dependencies. The fix is to run pip install kikit --force-reinstall. Pip will freshly resolve the dependency tree and KiKit should work again.

closed 2024-11-17 by yaqwsx

jj-uk, 2025-01-02:
@yaqwsx Thanks for the solution. Are you able to update the windows install guide, here:  https://yaqwsx.github.io/KiKit/v1.3/installation/windows/

May save some people some time in the future.