yaqwsx/KiKit #766: Some tabs do not generate with custom layout

JasperZP, 2024-11-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

`kikit, version 0+unknown` (installed extra/python-kikit 1.6.0-2 on archlinux)

### KiCAD version

8.0.6

### Operating system

Archlinux, 6.11.7

### Description

Some tabs do not generate with custom layout.
I wrote a simple layout plugin that offsets alternating rows by 1.6mm, and rotates the first row by 180deg which looks like this:
[image]

However, it does not generate all tabs as specified by the KiKit:tab footprint, especially those on the right side of the board:
[image]

If I use the same layout plugin but remove the horizontal offset, the tabs do generate correctly.
The partition lines and bouding boxes all seem to be correct.

### Steps to Reproduce

1. Download the attached board, config file, and plugin
1. Run: `kikit panelize -p kikit_config.json main_bare.kicad_pcb panel.kicad_pcb`
1. Check output to see that tabs are not generated on one side of the board
[files.zip](https://github.com/user-attachments/files/17798754/files.zip)

yaqwsx, 2024-11-18:
I just took your code and ran it - the partition line definitely doesn't look correct:

[image]

This is the reason why the tabs are not generated. I haven't been able to find the problem in the partition line generator so far. As a temporary workaround, I suggest building the partition line in the script manually. It should be straightforward for your layout.

JasperZP, 2024-11-19:
Ahh that looks more like the partition line I get. How were you able to show the partition lines like that? When I run `panel.debugRenderPartitionLines()` at the end of the layout plugin I get this on layer User.Eco1:
[image]

JasperZP, 2024-11-20:
Ah nevermind, I got it working now (when I ran `panel.debugRenderPartitionLines()` at the end of the `buildPartitionLine` function it shows the real parition line). Created the lines manually and now it generates the tabs correctly.

Viermusketiere, 2024-12-16:
I do have some more input on this issue.
I created a script to build a panel from a flexible amount of single- or multiboard projects.
When laying the isolated boards out into the panel the generation of the partition lines gives some unexpected and feature breaking results.
[image]
Markings of missing tabs:
[image]

When there is a continuous gap from left to right or top to bottom the generation of the partition line seems to be working just fine. When aligning multiple boards with different dimensions the generation of the partition line seems to work only on 2 of 4 sides (see board 4 and 5). When I move board 4 downwards to align with the bottom height of board 3 the tabs between board 4 and 5 are generated but the tabs between 2 and 4 are missing.
Another interesting point is the misoriented piece of partition line between board 3 and 4.
All board outlines of the source boards are continuous lines or polygons. There are no remarkable issues on the boards in the regions where partition lines are missing or malformed.

If my project files are required please let me know, as it will be some work to anonymize the boards and isolated the panel generation script from my build pipeline. But if it helps I will gladly provide the files!

@JasperZP How did you manage to manually build the partition lines?

JasperZP, 2024-12-16:
> @JasperZP How did you manage to manually build the partition lines?

I made a layout plugin which defines the layout and the partition lines (just hardcoded though, here's a summary):
```python
class LegacySGNPanelLayout(LayoutPlugin):
    def buildLayout(
        self, panel: Panel, inputFile: str, sourceArea: pcbnew.BOX2I
    ) -> Iterable[Substrate]:
        """
        This function is supposed to build the layout (append the boards to the
        panel) and return an iterable of substrates of these boards.
        """
        ...
        return panel.substrates

    def buildPartitionLine(
        self, panel: Panel, framingSubstrates: Iterable[Substrate]
    ) -> None:
        """
        This function should build the partition line in the panel. It gets an
        iterable of extra substrates that represent soon-to-be frame of the
        panel.
        """
        lines = []
        lines.append(LineString(coordinates))
        substrate.partitionLine = shapely.ops.linemerge(lines)
        panel.debugRenderPartitionLines()        
```