yaqwsx/KiKit #773: Kikit throws the following error: Cannot perform: Error in section layout: Cannot read quantity '5'

kaizen75, 2024-12-23:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.6

### Operating system

Windows x64

### Description

When using Kikit I get the error -  Cannot perform: Error in section layout: Cannot read quantity '5'

I played around with the values for the rows and columns but nothing works. Any help much appreciated!

### Steps to Reproduce

1) click KiKit icon
2) select input file
3) select output file
4) set vspace to 2mm
5) set hspace to 2mm
6) set rows to 5
7) set columns to 5
8) click panelize

yaqwsx, 2024-12-23:
I just tested it with the steps you provided and it works in my setup. Please, enable `--debug 'trace: true'` and post the output.

yaqwsx, 2025-01-26:
As there is no further activity, closing. If you provided further information on reproduction (e.g., the stack trace), feel free to reopen.

closed 2025-01-26 by yaqwsx