yaqwsx/KiKit #781: Not backend found popup despite it seems to be installed

ask6483, 2025-01-13:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.7+b1

### Operating system

Debian Trixie

### Description

Despite KiKit backend installed the 'KiKit installation not found!' message is displayed each time board file is opened in kicad.
Installation was working fine before Kicad update, I don't remember what the previous version was.

### Steps to Reproduce

Open the board file.

yaqwsx, 2025-01-14:
How do you verify that KiKit backend is installed? Could you open Scripting console (in PCB editor go to Tools -> Scripting editor) and enter:
```
from kikit.actionPlugins import importAllPlugins
importAllPlugins()
```

What errors do you get?

ask6483, 2025-01-14:
```
from kikit.actionPlugins import importAllPlugins
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'kikit'
importAllPlugins()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'importAllPlugins' is not defined
```

ask6483, 2025-01-14:
```
kikit --version
kikit, version 1.6.0
```

```
which kikit
/usr/local/bin/kikit
```

ask6483, 2025-01-14:
Result of the run in python console
```
Python 3.12.8 (main, Dec 13 2024, 13:19:48) [GCC 14.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from kikit.actionPlugins import importAllPlugins
>>> importAllPlugins()
./pcbnew/action_plugin.cpp(163): assert "PgmOrNull()" failed in register_action().
****************************************************************************
***                                                                      ***
*** Fatal error                                                          ***
*** The application handle was destroyed after running Python plugin ''. ***
***                                                                      ***
****************************************************************************
Aborted
```

yaqwsx, 2025-01-14:
The output says that you cannot import `kikit` from the Python interpreter that KiCAD uses. Hence, no backend is found. Is it possible that you have KiCAD setup in a way that it overrides `PYTHONPATH`?

ask6483, 2025-01-14:
Paths:

a. python console
```
>>> import sys
>>> print(sys.path)
['', '/usr/lib/python313.zip', '/usr/lib/python3.13', '/usr/lib/python3.13/lib-dynload', '/usr/local/lib/python3.13/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.13/dist-packages']
```

b. kicad console
```
import sys
print(sys.path)
['', '/usr/lib/python3/dist-packages', '/usr/lib/python313.zip', '/usr/lib/python3.13', '/usr/lib/python3.13/lib-dynload', '/usr/local/lib/python3.13/dist-packages', '/usr/lib/python3.13/dist-packages', '/usr/share/kicad/scripting', '/usr/share/kicad/scripting/plugins', '/home/ask/.local/share/kicad/8.0/scripting', '/home/ask/.local/share/kicad/8.0/scripting/plugins', '/home/ask/.local/share/kicad/8.0/3rdparty/plugins', '/usr/share/kicad/scripting']
```

ask6483, 2025-01-14:
I have switched python version using `update-alternatvies` to check if that helps but it's the same: only kikit is not working.

ask6483, 2025-01-14:
Managed to fix the issue by re-installing kikit with pip but by using the exact python interpreter registered in system `python -m pip install kikit`.
When installing using `pip install kikit` the package is installed to the older python interpeter version paths.
Seems that there is some kind of problem related to the transition to the new python version in Dabian testing from 3.12 to 3.13.

closed 2025-01-14 by ask6483

elpiel, 2025-01-19:
Same here. I used a separate GitHub action to install kikit and it was failing to find `pcbnew` despite KiCad being installed on the runner.

Use the default python version on the system to fix this issue.