yaqwsx/KiKit #783: Found and 'fixed' an issue

uggima, 2025-01-14:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

1.6.0

### KiCAD version

8.0.8 (although was the same in 8.0.6)

### Operating system

Windows

### Description

Gui and CLI both presented me with:

An error occurred: Expected '(', got ''

### Steps to Reproduce

Running any panelize command.

%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board>    "%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board\esp8266 wled board.kicad_pcb" "%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board\pannel esp8266 wled board.kicad_pcb" --debug trace:true

%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board>kikit panelize "esp8266 wled board.kicad_pcb" "pannel esp8266 wled board.kicad_pcb" --debug trace:true -p ..\..\kikit_panel3.json
An error occurred: Expected '(', got ''
No output files produced
Traceback (most recent call last):
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui.py", line 271, in doPanelization
    ki.buildLayout(preset, panel, input, sourceArea)
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize_ui_impl.py", line 257, in buildLayout
    substrates = panel.makeGrid(
                 ^^^^^^^^^^^^^^^
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 1331, in makeGrid
    boardSize = self.appendBoard(
                ^^^^^^^^^^^^^^^^^
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 1044, in appendBoard
    self._inheriCustomDrcRules(board, netRenamerFn)
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\panelize.py", line 818, in _inheriCustomDrcRules
    rules = parseSexprListF(f)
            ^^^^^^^^^^^^^^^^^^
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\sexpr.py", line 213, in parseSexprListF
    s = readSexpr(stream, limit=limit)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\sexpr.py", line 165, in readSexpr
    stream.shift("(")
  File "%userprofile%\Documents\KiCad\8.0\3rdparty\Python311\site-packages\kikit\sexpr.py", line 33, in shift
    raise ParseError(f"Expected '{expected}', got {repr(c)}")
kikit.sexpr.ParseError: Expected '(', got ''

%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board>python --version
Python 3.11.5

%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board>kikit --version
kikit, version 1.6.0

%userprofile%\Documents\Kicad\esp8266 wled\esp8266 wled board>pip install kikit
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: kikit in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (1.6.0)
Requirement already satisfied: click>=7.1 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (8.1.7)
Requirement already satisfied: commentjson>=0.9 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.0)
Requirement already satisfied: markdown2>=2.4 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.5.0)
Requirement already satisfied: numpy in c:\program files\kicad\8.0\bin\lib\site-packages (from kikit) (2.2.1)
Requirement already satisfied: pcbnewTransition<=0.5,>=0.4.1 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.4.2)
Requirement already satisfied: pybars3>=0.9 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (0.9.7)
Requirement already satisfied: shapely>=2.0.3 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (2.0.6)
Requirement already satisfied: solidpython>=1.1.2 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from kikit) (1.1.3)
Requirement already satisfied: colorama in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from click>=7.1->kikit) (0.4.6)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from pybars3>=0.9->kikit) (0.5.1)
Requirement already satisfied: PrettyTable==0.7.2 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.7.2)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: ply<4.0,>=3.11 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in %userprofile%\documents\kicad\8.0\3rdparty\python311\site-packages (from solidpython>=1.1.2->kikit) (0.0.19)

Solution was to remove 

    """
    Build layout for the boards - e.g., make a grid out of them.

    Return the list of created substrates and framing substrates. Also ensures
    that the partition line is build properly.
    """

from panelize_ui_impl.py and it works?

uggima, 2025-01-14:
And scratch the 'fixed' bit. couple of runs later and back to failing with not being able to reproduce a success, attempting with other files as well.

uggima, 2025-02-11:
Sorry for slow reply and lack of source file, attempted with another file at the time and it behaved the same. Now tried two other files and its working perfectly. Added .log to the  kicad_pcb to allow upload.

[esp8266 wled board.kicad_pcb.log](https://github.com/user-attachments/files/18750071/esp8266.wled.board.kicad_pcb.log)

kyhm, 2025-02-24:
@yaqwsx @uggima I encountered this today, my first time using kikit 1.7.0 to panelize a board designed in Kicad 8.0.9.  Looking at `_inheriCustomDrcRules` in the traceback hinted me that it's expecting the custom DRC rules to be non-empty. The board was a simple 2-layer breakout created with no custom DRC rules, and the workaround for me was to add this via gui under Board Setup -> Design Rules -> Custom Rules -> DRC rules:
```
(version 1)
```

Github won't let me attach the resulting `.kicad_dru` file, but the workaround changed it from empty to containing the above.

yaqwsx, 2025-02-24:
@kyhm Thank you for the investigation. This is the first time I have seen the file exist, but being empty. 

Fixed in 5c25a714f31d7808c0043ef01f2f2a2b4f387660.

closed 2025-02-24 by yaqwsx