yaqwsx/KiKit #788: Unable to correctly install KiKit (or Python ?)

PinheadBE1400, 2025-01-25:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

Unknown since installation fails

### KiCAD version

8.0.8 (8.0.7 = same result)

### Operating system

Windows 11

### Description

Previously, with KiCad 7 all worked well, but with Kicad 8, I cannot install nor run KiKit again.
The installation seems to work well, but when ruinning "kikit --help" it seems to try to use an old version of Python.
I tried completely remove (Uninstall with Windows) Kicad 8.0.8 and reinstalling it from scratch, then run "pip install kikit" from the Kicad Command Prompt, which didn't help when running (starting the pcb editor give also the message that kikit is not installed)
Note this was already the case with Kicad 8.0.7

[KiCad_8.0_Command_Prompt.txt](https://github.com/user-attachments/files/18545851/KiCad_8.0_Command_Prompt.txt)
[image]

Application: KiCad PCB Editor x64 on x64

Version: 8.0.8, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.0.1
	FontConfig 2.15.0
	libcurl/8.10.1-DEV Schannel zlib/1.3.1

Platform: Windows 11 (build 26100), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: NVIDIA Corporation, NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2, 4.6.0 NVIDIA 512.78

Build Info:
	Date: Jan 11 2025 18:59:49
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI

Build settings:

### Steps to Reproduce

1. Install Kicad 8.0.8 from scratch
2.  Open Kicad Command Prompt
3.  type 'pip install kikit'    (See attachment)
4. when completed type 'kikit --help'   ==> gives an error (apparently in python version invoked) (see attachment)

yaqwsx, 2025-01-26:
It seems that you got Python packages on your system in an inconsistent state. It seems that the `PYTHONPATH` and `PATH` environmental variables might refer to non-existent locations. This has to be probably fixed manually. If they are correct, I would try to enforce reinstallation of Kikit via `pip install --force-reinstall kikit`.

PinheadBE1400, 2025-01-26:
Hi, many thanks for your help!

- PATH did, indeed, contain a path to python39.bak which was uninstalled.   I corrected the path to point to python311 instead
- I then "pip install --force-reinstall kikit" and this solved the problem 

Suggestion: Including these checkpoints in the documentation may help other users

closed 2025-01-26 by yaqwsx