yaqwsx/KiKit #791: Plugin mechanism doesn't work

abetis, 2025-01-31:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.6.0 (older version are not supported)

### KiKit version

kikit, version 1.6.0

### KiCAD version

8.0.8

### Operating system

Ubuntu 22.04

### Description

Copied the plugin.py to the project local directory.
In JSON configuration I have this section:
"fiducials": {
"type": "plugin",
"code": "plugin.py.FiducialsPlugin"
}

But kikit throws this error:
An error occurred: Error in section fiducials: Invalid plugin type specified, FiducialsPlugin expected

I followed the plugins explanation page. plugin.py as it's available from: kikit/plugin.py
The file have the class "FiducialsPlugin".

Was expecting to see the "Fiducials plugin has to provide buildFiducials" exception from the default plugins.py file, but something fails before that.

### Steps to Reproduce

Copy the plugin.py to the project local directory.
In JSON configuration I have this section:
"fiducials": {
"type": "plugin",
"code": "plugin.py.FiducialsPlugin"
}

But kikit throws this error:
An error occurred: Error in section fiducials: Invalid plugin type specified, FiducialsPlugin expected

yaqwsx, 2025-01-31:
Since you haven't provided the code, it is hard to judge. But I guess your plugin doesn't inherit from a `FiducialsPlugin`.

abetis, 2025-01-31:
Can't I use the default FiducialsPlugin class implementation from the plugin.py just to see the "Fiducials plugin has to provide buildFiducials" error thrown in order to see that everything is set up correctly?

Is there an example with a simple class that inherit from the FiducialsPlugin class?
I'm not a Python person...