yaqwsx/KiKit #795: KiCAD 9.0.0 RC3 - "Object has no attribute" remains on Kikit 1.7.0 release

FractalEngineer, 2025-02-20:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.0 (older version are not supported)

### KiKit version

1.7.0

### KiCAD version

9.0.0 RC3

### Operating system

Win11

### Description

Just reporting on the error message, it appears to fail showing the preview on the PCBNew instance, but the output file comes out fine

### Steps to Reproduce

[image]

conradopoole, 2025-02-21:
I'm having the exact same issue, on the official 9.0 release

SmakyTaky, 2025-02-22:
Same here.. failing with the exact same message on the Official 9.0 release.

rtm7777, 2025-02-23:
Same for me on Mac

rtm7777, 2025-02-23:
https://github.com/yaqwsx/KiKit/blob/master/kikit/actionPlugins/panelize.py#L80

this is because of new version

riacob, 2025-02-23:
Same issue, need fixing asap as I have to order boards

yaqwsx, 2025-02-24:
This is embarrassing; I ran the latest tests on a nightly version as I wanted to be ready with KiKit before the official v9 release. Since the version is 8.99, it is classified as both v8 and v9, hence it passed my tests.

Fixed in 995fe49ba9b4109c3a09cc60b31fe3cf05204903. Release v 1.7.1 was made.

@riacob: In such cases, when time matters, it is always possible to reach out for paid support.

closed 2025-02-24 by yaqwsx