yaqwsx/KiKit #797: JLC fab output gerber layers missing and duplicated on a 6-layer board

mairas, 2025-02-27:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.1 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.7.1+2.g8ca9030

### KiCAD version (only KiCAD 8 and 9 are supported)

Application: KiCad Gerber Viewer arm64 on arm64  Version: 9.0.0, release build  Libraries: 	wxWidgets 3.2.6 	FreeType 2.13.3 	HarfBuzz 10.2.0 	FontConfig 2.15.0 	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.63.0  Platform: macOS Sequoia Version 15.3.1 (Build 24D70), 64 bit, Little endian, wxMac OpenGL: Apple, Apple M1 Max, 2.1 Metal - 89.3  Build Info: 	Date: Feb 19 2025 13:05:44 	wxWidgets: 3.2.6 (wchar_t,wx containers) 	Boost: 1.87.0 	OCC: 7.8.1 	Curl: 8.7.1 	ngspice: 44.2 	Compiler: Clang 16.0.0 with C++ ABI 1002 	KICAD_IPC_API=ON

### Operating system

MacOS Sequoia Version 15.3.1

### Description

I am creating JLC fabrication files for a six-layer board. In the resulting gerber files, top soldermask is duplicated to inner1.gts and the inner1 copper layer is missing. Likewise, bottom soldermask is duplicated to inner3.gbs and inner3 is missing.

### Steps to Reproduce

The project seems to be too big to upload but you can clone this repo: https://github.com/hatlabs/HALPI2-hardware and use this commit: 8f43056a4a90e2fb7dec9faa3e2117a4e0c2db4c

Then run this fabrication script (sorry, should've included that in the repo):

```
#!/usr/bin/env bash

set -euo pipefail
#shopt -s inherit_errexit

BOARD=HALPI2

# Create a 1x1 "panel" with top and bottom rails

echo "↔️ Panelizing..."

kikit panelize \
    --layout 'grid; rows: 1; cols: 1; space: 2mm' \
    --tabs 'fixed; vwidth: 55mm; hwidth:46.5mm; vcount: 1; hcount: 1' \
    --cuts 'vcuts; layer: Edge.Cuts' \
    --framing 'frame; width: 5mm; space: 2 mm' \
    --tooling '4hole; hoffset: 2.5mm; voffset: 2.5mm; size: 1.152mm' \
    --fiducials '4fid; hoffset: 5mm; voffset: 3.85mm; coppersize: 2mm; opening: 1mm;' \
    --text 'simple; text: JLCJLCJLCJLC; anchor: mt; voffset: 2.5mm; hoffset: 20mm; hjustify: center; vjustify: center;' \
    --post 'millradius: 1mm' \
    ${BOARD}.kicad_pcb ${BOARD}-panel.kicad_pcb

echo "🔩 Generating fabrication files..."

kikit fab jlcpcb \
    --no-drc \
    --assembly \
    --no-autoname \
    --schematic ${BOARD}.kicad_sch \
    ${BOARD}-panel.kicad_pcb \
    assembly-panel
```

Then compared the produced gerber files to the board layers.

mairas, 2025-02-27:
I added some printf debugging in the inner copper layer output loop in `export.py:gerberImpl()` and it would seem there's an off-by-one error wrt. inner layer indexing and what `pctl.SetLayer()` assumes:

```
plotting inner1, id 1
plot file /Users/mairas/w/hatlabs/HALPI/HALPI2-hardware/assembly-panel/gerber/HALPI2-panel-CmtUser.gbr
plotting inner2, id 2
plot file /Users/mairas/w/hatlabs/HALPI/HALPI2-hardware/assembly-panel/gerber/HALPI2-panel-inner1.gts
plotting inner3, id 3
plot file /Users/mairas/w/hatlabs/HALPI/HALPI2-hardware/assembly-panel/gerber/HALPI2-panel-inner2.gbl
plotting inner4, id 4
plot file /Users/mairas/w/hatlabs/HALPI/HALPI2-hardware/assembly-panel/gerber/HALPI2-panel-inner3.gbs
```

But then again, doing `pctl.SetLayer(innerlyr + 1)` doesn't change anything! The exact same output.

yaqwsx, 2025-02-27:
KiCAD changed layer numbering between v8 and v9. I just checked, and we haven't reflected this in the export code. I will look into that.

yaqwsx, 2025-02-27:
This should be fixed in 2954a480c2bd69f0921ab7cb588523078521cfc5. Please try the upstream version and confirm it works as expected.

mairas, 2025-02-27:
It's better. :-) But now there seems to be two extra copper layers (inner5 and inner6):

[image]

mairas, 2025-02-27:
"The two most difficult things in computer science are cache invalidation, naming things and off-by-one errors."

mairas, 2025-02-28:
On export.py line 138, you're now calling innerCu with the total layer count instead of just the inner layer count:

```
        for i, layer in enumerate(Layer.innerCu(board.GetCopperLayerCount() - 2)):
```

That change fixed it for me.

yaqwsx, 2025-02-28:
Fixed.

closed 2025-02-28 by yaqwsx

markh-de, 2025-03-03:
@yaqwsx: I'm still on 1.7.1 and having the same layer issue. What is the cleanest way to get the latest **upstream** version installed? Are there CI build artifact packages that can be installed with pip (as it seems to be the case with the PCM packages), or do I need to clone/copy the files over the 1.7.1 installation manually?

Thank you!