yaqwsx/KiKit #798: KiCAD9 AttributeError: 'NoneType' object has no attribute 'GetDrawings'

darthcloud, 2025-02-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.7.1 (older version are not supported)

### KiKit version

kikit, version 1.7.1

### KiCAD version (only KiCAD 8 and 9 are supported)

KiCAD 9

### Operating system

Kubuntu 24.04

### Description

I took one of my design done in KiCAD 7 originaly and uplated them to KiCAD 9 (Which was simply openening them in KiCAD 9 and saving basicly)

I had panelized that design with KiKit before and reused the exact same cmd:

```
kikit panelize     --layout 'hspace: 8mm; vspace: 2mm; hbackbone: 5mm; rows: 2; cols: 3; vbonecut: False'     --tabs 'type: annotation; tabfootprints: xbox:Tab'     --cuts 'type: mousebites; drill: 0.6mm; spacing: 0.9mm; offset: -0.3mm; prolong: 1mm'     --framing 'type: frame; mintotalheight: 70mm; mintotalwidth: 70mm; cuts: h; fillet: 1mm'     --tooling 'type: 4hole; hoffset: 2.5mm; voffset: 2.5mm'     --fiducials 'type: 4fid; hoffset: 12mm; voffset: 3.85mm; opening: 3mm'     --post 'millradius: 1mm'     'xbox.kicad_pcb' 'panel/panel' --debug 'trace: true'
An error occurred: 'NoneType' object has no attribute 'GetDrawings'
No output files produced
Traceback (most recent call last):
  File "/home/jacques/.local/lib/python3.12/site-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/home/jacques/.local/lib/python3.12/site-packages/kikit/panelize_ui.py", line 305, in doPanelization
    panel.save(reconstructArcs=preset["post"]["reconstructarcs"],
  File "/home/jacques/.local/lib/python3.12/site-packages/kikit/panelize.py", line 603, in save
    for edge in collectEdges(fillBoard, Layer.Edge_Cuts):
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jacques/.local/lib/python3.12/site-packages/kikit/common.py", line 73, in collectEdges
    for edge in chain(board.GetDrawings(), *[m.GraphicalItems() for m in board.GetFootprints()]):
                      ^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'GetDrawings'
```

### Steps to Reproduce

[pcb.zip](https://github.com/user-attachments/files/19017883/pcb.zip)

Use this board with the command below:
```
kikit panelize     --layout 'hspace: 8mm; vspace: 2mm; hbackbone: 5mm; rows: 2; cols: 3; vbonecut: False'     --tabs 'type: annotation; tabfootprints: xbox:Tab'     --cuts 'type: mousebites; drill: 0.6mm; spacing: 0.9mm; offset: -0.3mm; prolong: 1mm'     --framing 'type: frame; mintotalheight: 70mm; mintotalwidth: 70mm; cuts: h; fillet: 1mm'     --tooling 'type: 4hole; hoffset: 2.5mm; voffset: 2.5mm'     --fiducials 'type: 4fid; hoffset: 12mm; voffset: 3.85mm; opening: 3mm'     --post 'millradius: 1mm'     'xbox.kicad_pcb' 'panel/panel' --debug 'trace: true'
```

yaqwsx, 2025-02-28:
The error comes from the fact that you are trying to name your panel `panel` instead of `panel.kicad_pcb`. I added a check for this in 8b7a52b.

closed 2025-02-28 by yaqwsx

darthcloud, 2025-02-28:
Oh yeah that was it!

Thanks for the help and the awesome plugin!