yaqwsx/KiKit #799: V-Cuts not generating on some files

FractalEngineer, 2025-03-01:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.1 (older version are not supported)

### KiKit version

1.7.1

### KiCAD version (only KiCAD 8 and 9 are supported)

8.0.8

### Operating system

Win11

### Description

Hi,

Some of my jobs just seem to not want to print the V-cuts on the panelized output; not sure if it might be a size issue or something else?

Here are 2 examples of a working and bad one with the same config:

Good one (11.6 x 15.6mm)
[image]

Bad one (9.5 x 7.2mm)
[image]

### Steps to Reproduce

Please find the faulty file attached; thanks!

[filter.zip](https://github.com/user-attachments/files/19037997/filter.zip)

yaqwsx, 2025-03-02:
Cuts are only generated by tab placement. If you place your tabs such that they don't separate the PCB, there won't be any cuts (https://yaqwsx.github.io/KiKit/latest/panelization/tabs/).

In your second case the default tab size is smaller than the PCB, hence, no tab is placed, thus no cuts is placed. 

In you case you should choose full width tabs and everything will turn out as expected.

closed 2025-03-02 by yaqwsx

FractalEngineer, 2025-03-03:
Ah! that was it, I will remember that thank you very much :)