yaqwsx/KiKit #802: Add layout offset while keeping centered V-Cut line

FractalEngineer, 2025-03-05:
### Prerequisites

- [x] I have read the documentation and the proposed feature is not implemented.

### Description

Most manufacturers have different clearances for copper to milled and v-cut edge; for instance 0.2 and 0.4mm

I was looking to simply add 0.4mm offset in the layout to pass the V-cut requirement, but this appears to draw 2 lines, at the original board edge

[image]

Would it be possible to implement something as such?

I know there is already a clearance option reducing the copper to edge, I would like to go the other way and increase the board size to keep the copper areas as is

Thanks!

FractalEngineer, 2025-03-05:
Here's the expected result on the vertical cuts by manually deleting one set of v-cuts and moving the remaining one back to center

[image]

Also noticing the offset didn't apply to the rail; I've added some vspace and will be offsetting the cut manually as well

Such setting could be called "copper to vcut clearance" in the v-cut settings

FractalEngineer, 2025-03-05:
Here is what the desired output would look like; single, centered v-cuts with 0.4mm offset between every V-cut line and copper area

[image]

Devilbinder, 2025-04-20:
Second this. Board spacing does not adhere to the `clearance` parameter when using `vcuts`. It merely draws a keep out area around the V-CUT.
The board spacing appears to be fixed to be 0.2mm from the  V-CUT center line. Making this an adjustable parameter would be appreciated. 

My current work around to set `xbackbone` to the desired spacing then manually delete the duplicate V-CUT lines and move the remaining V-CUT lines to the correct position.

Generated output:
[image]

Edited:
[image]

Settings used
```json
{
    "layout": {
        "alternation": "rows",
        "cols": "2",
        "rows":"1",
        "vbackbone": "0.4mm"
    },
    "cuts": {
        "type": "vcuts",
        "clearance": "0.8mm",
        "layer": "Edge.Cuts"
    },
    "framing": {
        "type": "frame",
        "hspace": "0mm",
        "vspace": "0mm",
        "width": "4mm",
	"thickness": "0mm",
	"mintotalheight":"70mm",
	"mintotalwidth":"70mm"
    },
    "tabs": {
    "type": "full",
    "vcount": 2,
    "hcount": 2,
    "width": "3mm"
  }
}

```