yaqwsx/KiKit #810: 4 layer gerbers fail

VisaJE, 2025-03-14:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.1 (older version are not supported)

### KiKit version

kikit, version 0+unknown

### KiCAD version (only KiCAD 8 and 9 are supported)

9.0.0

### Operating system

6.13.6-arch1-1

### Description

Running the command `kikit fab jlcpcb *.kicad_pcb --assembly --schematic *.kicad_sch gerbers/rev1` produces a gerber zip without warnings (other than missing order codes), but the gerbers do not contain the correct copper layers.

The gerber viewer shows 3 copper layers, one of which is a copy of another one, so the two inner layers are missing.

Since our project is done with Kicad 9 it is not easy to migrate to an older version. Furthermore, trying the docker container for kikit did not work for a Kicad 9 project. We have not been able to test other versions  of kikit.

Attached are the problematic gerbers.
[gerbers.zip](https://github.com/user-attachments/files/19245516/gerbers.zip)

### Steps to Reproduce

[issue-project.zip](https://github.com/user-attachments/files/19245522/issue-project.zip)
1. Unzip the issue project
2. Run `kikit fab jlcpcb *.kicad_pcb --assembly --schematic *.kicad_sch gerbers/rev1`
3. Observe the generated output for example with kicad gerber viewer

VisaJE, 2025-03-14:
The version code being unknown is obviously rather suspicious,  but I installed kikit through the arch package system and there it says `extra/python-kikit 1.7.1-1` which makes me believe the version is 1.7.1.

VisaJE, 2025-03-14:
Additional piece of information is that Arch linux is now sporting python 3.13.2, which I believe the system kicad would also be using. Installing the kikit 1.7.1 in a virtual environment is not working: I get import issues  trying to use it.

VisaJE, 2025-03-14:
More information: I replicated the issue on kicad docker image. Required a bit of work to install kikit in a virtual env but there it says `kikit, version 1.7.1` for `kikit --version`

yaqwsx, 2025-03-14:
I bet this is a duplicate of #797 (the latest closed issue). Could you confirm this?

VisaJE, 2025-03-14:
Confirmed : )
The tool works as expected on the newest commit.

closed 2025-03-14 by VisaJE

yaqwsx, 2025-03-14:
Note that you should really do what the checkmarks in the issue template say before submitting an issue. You could have resolved your issue 2 hours sooner.