yaqwsx/KiKit #816: An error occurred: 'NoneType' object has no attribute 'GetDesignSettings'

greenscreenflicker, 2025-04-16:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.2 (older version are not supported)

### KiKit version

kikit, version 1.7.1

### KiCAD version (only KiCAD 8 and 9 are supported)

Kicad 9

### Operating system

Windows

### Description

I get the following error:

kikit panelize  --layout "grid; rows: 1; cols: 7; space: 0mm; rotation: 0deg;"  --tabs "full" --cuts "vcuts" --post "millradius: 0.3mm"   --source "tolerance: 20mm"  ../dcdcpower.kicad_pcb panel.kicad_pcb
An error occurred: 'NoneType' object has no attribute 'GetDesignSettings'
No output files produced

### Steps to Reproduce

run above command with the pcb provided by email

greenscreenflicker, 2025-04-16:
... turns out i was in the wrong file path. It should warn that file is not found; instead the message is misleading

yaqwsx, 2025-04-17:
Fixed in 82b8409.

closed 2025-04-17 by yaqwsx

yaqwsx, 2025-12-28:
Please, provide the stacktrace.