yaqwsx/KiKit #827: Round tab generation issue

tmandys, 2025-05-14:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.2 (older version are not supported)

### KiKit version

1.7.2

### KiCAD version (only KiCAD 8 and 9 are supported)

9

### Operating system

Windows 11

### Description

It is not possible to create rounded tabs because when fillet is not 0 then exception is raised. Rectangular tabs works well. I run standalone script to panelize irregular layout,

```
gap_width = 5*mm
mb_width = 2.4 * mm

 t, c = panel.boardSubstrate.tab(bite[0], bite[1], mb_width, maxHeight=gap_width, fillet=gap_width/2)
```

```
Traceback (most recent call last):
  File "panelize-boards.py", line 193, in <module>
    t, c = panel.boardSubstrate.tab(bite[0], bite[1], mb_width, maxHeight=gap_width, fillet=gap_width/2)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\Documents\KiCad\9.0\3rdparty\Python311\site-packages\kikit\substrate.py", line 868, in tab
    return self._makeTabFillet(tab, tabFace, fillet)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\Documents\KiCad\9.0\3rdparty\Python311\site-packages\kikit\substrate.py", line 942, in _makeTabFillet
    newFace = candidate.intersection(self.substrates.exterior)
                                     ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'MultiPolygon' object has no attribute 'exterior'
```

I suppose issue is:

-        newFace = candidate.intersection(self.substrates.exterior)
+       newFace = candidate.intersection(self.exterior)

But even in this case it does not work as newFace is generated as non LineString type and result is>

`raise TabFilletError(f"Unexpected result of filleted tab face:  <class 'shapely.geometry.multipolygon.MultiPolygon'>, {
    "type": "MultiPolygon",
    "coordinates": [
        [
            [
                [
                    6.1119166502845985,
                    46600999.981322065
                ],
                [
                    103.7178662832376,
                    46600994.60675858
                ],
                [
                    200.33270796864755,
                    46600979.72792454
                ],
                [
`
Polygon is not correct, not sure if there is an issue with rounding and it confuses intersection algorithm but works without fillets.

The first step is confirming if code fix is ok. In the second I could prepare a simplified kicad_pcb to reproduce

### Steps to Reproduce

The first step is confirming if code fix is ok. In the second I could prepare a simplified kicad_pcb, plus .py script to reproduce.

yaqwsx, 2025-05-14:
Based on the error you reported I can only say "some geometrical assert about the substrates has been violated". Without a reproducer I am not able to say more.

tmandys, 2025-05-14:
I understand it, but issue has two parts. The first is obvious (exteriour runtime error) so I need be sure my fix is ok and is not reason of second issue to proceed.

yaqwsx, 2025-05-15:
I don't understand why the exterior error is obvious.

tmandys, 2025-05-15:
IMHO you implemented `panel.exterior()` wrapper method to cope with this kind of issue when `exterior` field is missing in some polygon classes.