yaqwsx/KiKit #830: "No board edges found" after separation. Shapes turn to segments

JavanPoirier, 2025-05-28:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.2 (older version are not supported)

### KiKit version

1.7.2

### KiCAD version (only KiCAD 8 and 9 are supported)

9.0.2

### Operating system

Windows

### Description

I have a multiboard setup and am building a panelization script, and upon attempting to appendBoard, I get "No board edges found". 
```
    motherboard = panel.appendBoard(
        "boards/motherboard.kicad_pcb",
        [0, 0],
        origin=Origin.Center,
        rotationAngle=EDA_ANGLE(0, DEGREES_T)
    )
```

I discovered when opening the separated pcb file that my rectangle or polygon edge cuts of my board were made into segments after separation. 

```
cwd = os.getcwd()
input_file = os.path.abspath("vowloc.kicad_pcb")
os.makedirs("boards", exist_ok=True)

board_map = {
    "B1": "Motherboard.kicad_pcb",
    "B2": "Interface.kicad_pcb",
    "B3": "USB_Left.kicad_pcb",
    "B4": "USB_Right.kicad_pcb",
}

for ref, out_name in board_map.items():
    output_file = os.path.abspath(os.path.join("boards", out_name))

    # Escape inner quotes for cmd.exe
    command = (
        f'"{kicad_cmd}" && kikit separate --source "annotation; ref: {ref}" '
        f'"{input_file}" "{output_file}"'
    )

    # Wrap whole thing in one string to pass to cmd.exe
    shell_command = f'cmd.exe /c "{command}"'

    print(f"\nRunning: {shell_command}")
    result = subprocess.run(shell_command, check=True)
```

If I manually replace the segment edge cuts with rectangle or polygon edge cuts the appendBoard call is successful.

### Steps to Reproduce

1. Create a multiboard project
2. Seperate each board 
3. appendBoards to panel

yaqwsx, 2025-05-28:
Yes, KiKit internally converts everything to segments by design. It shouldn't, however, break the panelization. Your design must trigger some other problem. However, you haven't provided files for reproduction so I can only guess.

JavanPoirier, 2025-05-28:
Thanks for the reply! Attached is an example with 2 of my 4 boards, with my WIP scripts. I also noticed that on my USB board which uses a polygon edge cut does not show the gnd pour. The rectangle edge cut used on the Interface board does.

[example.zip](https://github.com/user-attachments/files/20482002/example.zip)