yaqwsx/KiKit #834: Missing pads and traces after panelization

efogdev, 2025-06-16:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.2 (older version are not supported)

### KiKit version

1.7.2

### KiCAD version (only KiCAD 8 and 9 are supported)

9.0.2 release build

### Operating system

Arch Linux

### Description

Original PCB:
[image]

After panelization:
[image]

Some traces and PTH holes are missing completely.

### Steps to Reproduce

Not sure how to reproduce with a new PCB. See this board: [repo](https://github.com/efogdev/nrf54l15-module/tree/6-layer)

Also here's my panelization config:
```json
{
    "layout": {
        "hspace": "2mm",
        "vspace": "1.6mm",
        "rows": "3",
        "cols": "4",
        "renamenet": "B{n}-{orig}"
    },
    "source": {
        "tolerance": "0.5mm"
    },
    "tabs": {
        "type": "corner",
        "width": "1.075mm",
        "fillet": "0.5mm"
    },
    "cuts": {
        "type": "mousebites",
        "drill": "0.4mm",
        "spacing": "0.55mm"
    },
    "framing": {
        "type": "railslr",
        "width": "6mm",
        "fillet": "1.5mm"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "4.5mm",
        "voffset": "3.5mm",
        "opening": "2mm"
    },
    "page": {
        "anchor": "tl",
        "posx": "0mm",
        "posy": "0mm"
    }
}
```

efogdev, 2025-06-16:
Upd: solved by increasing tolerance. Weird though.

yaqwsx, 2025-06-16:
There is nothing weird about it. As you read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/), you know that components are selected whether their center is placed inside the board or not (possibly enlarged by the tolerance parameter). 

Every catellated hole in your design is, unconventionally, a single footprint. You have them on quite random coordinates - e.g., your edges are at `Y=45,5754`, some of the footprints are at `Y=45,597628`, some of them are at `Y=45,619148` - which means they are outside the board area and hence, they are not selected for panelization.