yaqwsx/KiKit #854: An error occurred: Discontinuous outline at [138.1386, 28.8053] creating stencile from kikit panelized file

martin323232, 2025-08-24:
### Prerequisites

- [x] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [x] I have searched existing issues (including closed ones)
- [x] I use KiKit at least version 1.7.2 (older version are not supported)

### KiKit version

kikit, version 1.7.2

### KiCAD version (only KiCAD 8 and 9 are supported)

9.0.4

### Operating system

Windows 11

### Description

Trying to generate stencil from an existing panel created with kikit. The pcb board i round and its made into a square with kikit panelizing to be able to put it in PnP machine. All the "Discontinuous outline" errors is on the circle created with kikit panelizing. 

Same error on a m.2 board created with rails.

Works fine creating stencil on original file without rails.

### Steps to Reproduce

kikit stencil create  --jigsize 60 60 INPUTBOARD OUTPUTDIR

Output:

An error occurred: Discontinuous outline at [158.4684, 51.3223]. This may have several causes:
    - The outline in really discontinuous. Check the coordinates in your source board.
    - You haven't included all the outlines or in the case of multi-design,
      you have included a part of outline from a neighboring board.
No output files produced

yaqwsx, 2025-08-25:
You have not provided the required inputs - precise instructions to reproduce, including **input files** and **precise commands**.

I tried to reproduce your issue, but given the description, I failed. Until I can reproduce the issue, there is not much I can do. My test setup was:
- a file with a single circular PCB ([input.zip](https://github.com/user-attachments/files/21961464/input.zip))
- panelized with `kikit panelize /tmp/test.kicad_pcb /tmp/panel.kicad_pcb`
- stencil created with `kikit stencil create --jigsize 60 60 /tmp/panel.kicad_pcb /tmp/stencil`

yaqwsx, 2025-09-16:
Thank you for providing the files. They helped me to debug a hard-to-trigger condition. The problem is now solved in 29f6f76cbbb4b6fc66dfe13928f66e162102f8f7.

closed 2025-09-16 by yaqwsx